ros2/rosbag2 #912: Add option to recorder to ignore "leaf" topics

ivanpauno, 2021-11-17:
## Description

There're currently some options that allow the recorder to automatically figure out what topics to record: the "all" option and an "exclude" regex.
The proposal here is to add an option that will also exclude "leaf" topics (i.e. topic with no subscriptions).

For that we would need an extra filter, that probably checks the result of `get_publishers_info_by_topic()`.

## Related Issues

None

## Completion Criteria

Being able to run `ros2 bag record --all --exclude-leaf-topics`.

## Implementation Notes / Suggestions

We will need to look [here](https://github.com/ros2/rosbag2/blob/891e08128e2d6ff36452871abda2cd776b8a1566/rosbag2_transport/src/rosbag2_transport/topic_filter.cpp#L95-L121).

## Testing Notes / Suggestions

Probably extend https://github.com/ros2/rosbag2/blob/891e08128e2d6ff36452871abda2cd776b8a1566/rosbag2_transport/test/rosbag2_transport/test_record_all.cpp.

emersonknapp, 2021-11-24:
Is part of the motivation here to avoid activating "lazy publishers" - which won't actually publish messages unless a subscriber exists?

ivanpauno, 2021-11-24:
> Is part of the motivation here to avoid activating "lazy publishers" - which won't actually publish messages unless a subscriber exists?

Yes (all publishers are "lazy publishers" as nothing is sent if there are no subscriptions, even if multicast is enabled).
In a lot of cases you don't need to record "leaf topics", so it's a good idea to not "activate" their publishers to disturb the original system as less as possible.

emersonknapp, 2021-11-24:
> all publishers are "lazy publishers" as nothing is sent if there are no subscriptions, even if multicast is enabled

I was thinking more, at the application level, some publishers won't do any of the work to even construct messages and call `publish`, unless a subscription is active on their topic. As opposed to what you've said, which is at the middleware level. Both cases benefit from this feature though.

ivanpauno, 2021-11-24:
> Both cases benefit from this feature though.

Yes!
If this seems like a reasonable addition to `ros2 bag record` let me know and I might have time to work on it.

emersonknapp, 2021-11-24:
Yes, definitely this seems reasonable. For the implementation - I think high level this is a new `RecordOptions` option - and that the new logic would then exist somewhere under `Recorder::topic_discovery`

closed 2021-11-30 by gbiggs