ros2/rosbag2 #914: ros2 bag play option to enable LoanedMessage

fujitatomoya, 2021-11-23:
We would like to open discussion for this feature request.

## Description

Currently rosbag2 play publishes serialized messages, which is efficient to publish the messages from bag database w/o de-serialization.

https://github.com/ros2/rosbag2/blob/4792327ee6962152fa8f0f6b811e68a76c755d66/rosbag2_transport/src/rosbag2_transport/player.cpp#L485-L494

But if rmw implementation supports [LoanedMessage](https://design.ros2.org/articles/zero_copy.html), it can improve more performance when subscriptions are inside the same system of rosbag2 publisher. `rmw_cyclonedds` and `rmw_fastrtps`, both supports [LoanedMessage](https://design.ros2.org/articles/zero_copy.html) to achieve zero copy in the localhost.

## Implementation Notes / Suggestions

for doing this, 1st of all we need to discuss on either way to take.

- adding publish methods that allows us to publish serialized message with `LoanedMessage`. Currently there is no such interface to do so, so we need to add the method in `Publisher` class. this requires adding new interface in RMW and de-serialization always happens in subscription side even with `LoanedMessage` in localhost.
- de-serialize the message in `rosbag2 play` before publish, and then issue publish method with `LoanedMessage`. de-serialization only happens on `rosbag2 play` and then everything will be taken care by `LoanedMessage` to publish the messages via shared memory w/o serialization or encapsulation. (means no de-serialization is required on all subscription.)

I would like to take the latter one since it sounds most effective for this feature.

- `ros2 bag play` option to enable `LoanedMessage` publisher mode. (default disable)
- non-POD (Plain Old Data / Bound) or unbound data type, they needs to be published as serialized message (current behavior) w/o `LoanedMessage`, since these message types cannot be delivered via zero copy shared memory transport for now. 

## Testing Notes / Suggestions

- all regression test required.
- add `LoanedMessage` mode tests in `ros2 bag play`. (probably `rmw_fastrtps` would be suitable, since `rmw_cyclonedds` with `iceoryx` requires daemon process.)

fujitatomoya, 2021-11-23:
@emersonknapp we would like to open discussion here. i was going to talk about this last WG but didnt have chance. so probably next time, we could discuss on this? taking advantage of `LoanedMessage`, i guess it can improve the performance of `ros2 bag play`.

emersonknapp, 2021-11-23:
Sure - I added it to the agenda for the 12/3 meeting https://docs.google.com/document/d/1UXWaEkT6ji0Z0uTtJJSni5Y6fiopM9qOOzv4ldeNaAM/edit#

MichaelOrlov, 2021-11-30:
@fujitatomoya We have successfully implemented support for zero-copy in `Rosbag2` player internally in `ApexOS` and have tried to upstream it. Please refer to the issue #652 and relevant PR https://github.com/ros2/rosbag2/pull/664.  
It seems that we were running ahead of time and mainstream. 
May be it's time to reiterate and reopen those issue. 
Also please note that since then `LoanedMessage` has moved to the `rclcpp` and we will need to bring new API there first.

cc: @albtam

fujitatomoya, 2021-12-01:
@MichaelOrlov sounds really nice 👍 would you consider the PR against the OSS mainline? that would be really appreciated.

emersonknapp, 2021-12-03:
Discussed in Tooling WG, decided that it makes sense to use LoanedMessage, deserialize on the publishing side. May need to add utilities to rclcpp::GenericPublisher.

fujitatomoya, 2021-12-03:
@Barry-Xu-2018 @iuhilnehc-ynos we are good to go.

Barry-Xu-2018, 2021-12-06:
@emersonknapp  

> Discussed in Tooling WG, decided that it makes sense to use LoanedMessage, deserialize on the publishing side. May need to add utilities to rclcpp::GenericPublisher.  

Do you mean use existing class `LoanedMessage` ?   
https://github.com/ros2/rclcpp/blob/master/rclcpp/include/rclcpp/loaned_message.hpp#L32

If we want to use `LoanedMessage` in rclcpp::GenericPublisher, it seems that we need to change `LoanedMessage`.  
```c++
template<typename MessageT, typename AllocatorT = std::allocator<void>>
class LoanedMessage
```
For rclcpp::GenericPublisher, we cannot get message type.    

@emersonknapp  @fujitatomoya 
There is a new way to avoid modifying rosbag2.  
We directly update rclcpp::GenericPublisher::publish(const rclcpp::SerializedMessage & message).  
In this function, loan message,  deserialized and send message if publisher supports loan message.  

What do you think ?

Barry-Xu-2018, 2021-12-07:
> I believe that this is the approach that Apex.OS tried on here,  using and relying on rcl interfaces seem to be okay to me to address this feature.  

Yes. I considered only using rclcpp layer to send loaned messages. It will involve 2 types of publisher --- rclcpp::GenericPublisher and rclcpp::Publisher. This leads codes to become complex. So using rcl layer is better than rclcpp layer while borrowing/sending loaned message.    

> - instead of updating rclcpp::GenericPublisher::publish, why dont we have rclcpp::GenericPublisher::try_publish_loaned_msg
> - then, make an option rosbag2 play cli (default disable) to try publish via loaned message, if user wants to do so?

Completely agree. User decides it.

MichaelOrlov, 2021-12-07:
@fujitatomoya @Barry-Xu-2018 I want to clarify.
In ApexOS we didn't do
> We directly update rclcpp::GenericPublisher::publish(const rclcpp::SerializedMessage & message).
In this function, loan message, deserialized and send message if publisher supports loan message.

Instead we add a new API `GenericPublisher::publish_loaned_message(void * ros_message)` here is [original implementation](https://github.com/ros2/rosbag2/pull/664/commits/06b276f3dc030349a14fc41c8977c10f1ca57e79#diff-23ead5e1be60ecb3394cd912db08d0bff210eef95f99f2cfa87f008af4775e56R52-R70)

Barry-Xu-2018, 2021-12-07:
@MichaelOrlov 

In original implementation, `GenericPublisher` is in rosbag2. It was moved to rclcpp layer (rclcpp::GenericPublisher).  

> We directly update rclcpp::GenericPublisher::publish(const rclcpp::SerializedMessage & message).  

So we modify the same class `GenericPublisher`.  
About above consideration, I want to combine `GenericPublisher::publish_loaned_message` (mentioned by you) to rclcpp::GenericPublisher::publish().  After the discussion, new function is better. So we want to add new function rclcpp::GenericPublisher::try_publish_loaned_msg (Only the function name is different from GenericPublisher::publish_loaned_message). I create a new issue ros2/rclcpp#1837 in rclcpp to discuss whether rclcpp::GenericPublisher can be added a new function for loaned message (function name will be discussed in that issue).  

I completely agree with the implementation in [original implementation](https://github.com/ros2/rosbag2/pull/664/commits/06b276f3dc030349a14fc41c8977c10f1ca57e79#diff-23ead5e1be60ecb3394cd912db08d0bff210eef95f99f2cfa87f008af4775e56R52-R70).    
If rclcpp::GenericPublisher can add a new function for loaned message, the original implementation required only minor modifications.

fujitatomoya, 2022-03-28:
@Barry-Xu-2018 

I think we can proceed this issue based on rclcpp interfaces https://github.com/ros2/rclcpp/pull/1856.

Barry-Xu-2018, 2022-03-28:
@fujitatomoya 
 
> I think we can proceed this issue based on rclcpp interfaces [ros2/rclcpp#1856](https://github.com/ros2/rclcpp/pull/1856).

Yes, I start to make PR for rosbag2.

MichaelOrlov, 2022-05-10:
Closing as done in #981

closed 2022-05-10 by MichaelOrlov