ros2/rosbag2 #915: handle recording of messages larger than the sqlite blob limit (default is 1GB)

wjwwood, 2021-11-23:
## Description

Currently if you try to record a message larger than the sqlite blob/string limit, you get an exception and rosbag2 crashes due to an unhandled exception.

The default for sqlite (compile time configuration) is 1GB: https://sqlite.org/limits.html (`SQLITE_MAX_LENGTH`)

At the very least it should fail more gracefully, with an error or warning that tells you which topic the message came from and other details like the size of the message that was unsuccessfully written. I'm open to suggestion on what the behavior should be, but I'm leaning towards a warning that the message was not recorded (maybe with a warning "once per topic" or "throttled to once a second" or something), but continuing to record otherwise. I could see the argument for just stopping when this happens, but with a better error message.

Ideally, we'd extend rosbag2 so that it could split arbitrarily large messages across multiple database rows (pagination) as needed to avoid the blob limit.

## Related Issues

I ran into this working with a customer, and there's not a good work around other than avoiding the offending topic.

I'm working on a pull request to add a regression test and improve the error message, but I will not have time to implement pagination or anything like that.

## Completion Criteria

I think this issue should remain open until arbitrarily large messages can be recorded or until the maintainers decide to not support that use case.

## Implementation Notes / Suggestions

As I mentioned, the only idea I had for solving this would be to introduce pagination of large messages over multiple rows. I'm not sure what the implications are for the storage API or the "schema" for sqlite based rosbags.

## Testing Notes / Suggestions

Like I mentioned, I'm working on a regression test for this, but you should be able to trigger it by publishing anything over 1GB and recording it to trigger it.

emersonknapp, 2021-11-24:
Good ticket. Not sure if I'll have any resources to work on this - but it makes sense to me that rosbag2 should be able to support this use case. 

The one workaround I can think of off the bat would be to paginate the payload across multiple messages, rather than publishing a single payload as a 1GB+ individual message.

wjwwood, 2021-11-24:
> The one workaround I can think of off the bat would be to paginate the payload across multiple messages, rather than publishing a single payload as a 1GB+ individual message.

I think the message in question was part of the nav stack visualization using a Marker message, so maybe that could be an option, but changing that is probably not worth doing at this point. Probably just exclude it for now, or if we make it a warning then just ignore it.

emersonknapp, 2021-11-24:
Yes, I think warning/ignore is acceptable behavior. Definitely crashing is not OK. I believe this would be part of the storage implementation - and probably not part of generic functionality, considering that this particular limitaton is in SQLite. 

We could make that a note in the storage API that it should gracefully handle arbitrarily sized write - maybe have a return value for a successul/failed write instead of current `void BaseWriteInterface::write`

gbiggs, 2021-11-24:
I think that with the rapid ongoing development of a new bag file format, which should not have this problem, the best solution is to fail gracefully and note this as a limitation of sqlite3-format bag files. Unless someone has time to implement pagination within the sqlite3 storage plugin so that it's transparent to the rest of rosbag2.

emersonknapp, 2021-11-24:
This may be a bit tangential, but I'd be inclined to move the sqlite implementation into a package named `rosbag2_storage_sqlite3`, removing the package `rosbag2_storage_default_plugins` - instead indicating the default via `exec_depend` in the `rosbag2` metapackage. This sqlite3 implementation could then theoretically live in a separate repository. Not necessary but at least with that naming it would be a little easier to indicate "this is a problem in the storage implementation" vs "this is an API problem at the generic level". 

More on-topic, i think that allowing for graceful failing may only require the modification to the base interface to allow returning an indication of success/failure. Currently the API basically dictates that all write operations must succeed.

wjwwood, 2021-11-24:
I opened https://github.com/ros2/rosbag2/pull/919 to implement the warning behavior.

MichaelOrlov, 2023-03-17:
- Closing this as done. Since we have add warning instead of crash in SQLite3 backend and MCAP file format immune to the issue.  
- We are not going to support "pagging" for SQLite3 backend since new more efficient MCAP format available and uses by default.

closed 2023-03-17 by MichaelOrlov