ros2/rosbag2 #922: Some parameter events are not recorded

clarkebowers, 2021-11-29:
## Description
Some of the /parameter_events messages are not recorded in the rosbag2.

### Expected Behavior
All parameter events will be recorded in the ros bag.

### Actual Behavior
A random subset of the parameter events are recorded in the ros bag.

## To Reproduce
** Steps to reproduce the behavior, e.g.
1. Start bag recording in one terminal session: 

> ros2 bag record --all --include-hidden-topics

2. In another terminal session launch five instances of turtlesim_node using a launch file (see below). The launch file sets the three background color parameters for each instance.

> ros2 launch turtlesims.launch

3. Stop the five turtle sims.
4. Stop the ros bag capture.
4. Run the python script to dump the contents of the ros bag for parameter events (see script below). Use the name of the generated ros bag.

> python3 bagParams.py rosbag2_202...

Processing ros2bagrosbag2_2021_11_29-18_05_51/rosbag2_2021_11_29-18_05_51_0.db3
Found 24 parameter events in bag
[[1638209166.56, '/node4', 'New', 'use_sim_time', False],
 [1638209166.56, '/node4', 'New', 'background_r', 10],
 [1638209166.56, '/node4', 'New', 'background_g', 11],
 [1638209166.56, '/node4', 'New', 'background_b', 12],
 [1638209166.55, '/node2', 'New', 'use_sim_time', False],
 [1638209166.55, '/node2', 'New', 'background_r', 4],
 [1638209166.56, '/node2', 'New', 'background_g', 5],
 [1638209166.56, '/node2', 'New', 'background_b', 6],
 [1638209166.58, '/node1', 'New', 'use_sim_time', False],
 [1638209166.58, '/node1', 'New', 'background_r', 1],
 [1638209166.58, '/node1', 'New', 'background_g', 2],
 [1638209166.58, '/node1', 'New', 'background_b', 3]]

4. Examine the output. You will find less than all five turtlesim_node instances. Above you will only find three instances.
5. Repeat and you will find a different set of captured parameter events.

## System (please complete the following information)
 - OS: Ubuntu 20.04
 - ROS 2 Distro: Galactic
 - Version: release

## Additional context
This is the launch file: turtlesims.launch

<launch>
    <node pkg="turtlesim" exec="turtlesim_node" name="node1">
      <param name="background_r" value="1" />
      <param name="background_g" value="2" />
      <param name="background_b" value="3" />
    </node>
    <node pkg="turtlesim" exec="turtlesim_node" name="node2">
      <param name="background_r" value="4" />
      <param name="background_g" value="5" />
      <param name="background_b" value="6" />
    </node>
    <node pkg="turtlesim" exec="turtlesim_node" name="node3">
      <param name="background_r" value="7" />
      <param name="background_g" value="8" />
      <param name="background_b" value="9" />
    </node>
    <node pkg="turtlesim" exec="turtlesim_node" name="node4">
      <param name="background_r" value="10" />
      <param name="background_g" value="11" />
      <param name="background_b" value="12" />
    </node>
    <node pkg="turtlesim" exec="turtlesim_node" name="node5">
      <param name="background_r" value="13" />
      <param name="background_g" value="14" />
      <param name="background_b" value="15" />
    </node>
</launch>

#This is the python script to view the parameter events in the bag: bagParams.py
from pprint import pprint
import sys
import os
import sqlite3

from types import SimpleNamespace
from datetime import datetime
from pprint import pprint
from rosidl_runtime_py.utilities import get_message
from rclpy.serialization import deserialize_message

#Fetch all the scalar messages
class BagFileParser():
    def __init__(self, bag_file):
        self.conn = sqlite3.connect(bag_file)
        self.cursor = self.conn.cursor()

    def __del__(self):
        self.conn.close()

    def findBagFile(baglocation):
        if os.path.isdir(baglocation):
            for file_name in os.listdir(baglocation):
                if file_name.endswith(".db3"):
                    return os.path.join(baglocation, file_name)
        else:
            return baglocation 

    def fetch_parameter_events(self):
        ## create a message type map
        topics_data = self.cursor.execute("SELECT type FROM topics WHERE name = '/parameter_events' AND serialization_format = 'cdr' ORDER BY name").fetchone()
        topic_msg = get_message(topics_data[0])

        # Get from the db
        rows = self.cursor.execute("SELECT data FROM messages WHERE topic_id = (SELECT id from topics WHERE name = '/parameter_events') order by timestamp asc").fetchall()
        # Deserialise all scalar messages
        ret = [ deserialize_message(r[0], topic_msg) for r in rows ]
        return ret

#Parameter Report
class Parameters():
    def prepare(self):
        self.csv_rows = []
    
    def get_parameter_value(self, parameter_value):
        if parameter_value.type == 0:
            return ["n/a"]
        elif parameter_value.type == 1:
            return [parameter_value.bool_value]
        elif parameter_value.type == 2:
            return [parameter_value.integer_value]
        elif parameter_value.type == 3:
            return [parameter_value.double_value]
        elif parameter_value.type == 4:
            return [parameter_value.string_value]
        elif parameter_value.type == 5:
            return parameter_value.byte_array_value
        elif parameter_value.type == 6:
            return parameter_value.bool_array_value
        elif parameter_value.type == 7:
            return parameter_value.integer_array_value
        elif parameter_value.type == 8:
            return parameter_value.double_array_value
        elif parameter_value.type == 9:
            return parameter_value.string_array_value

    def add_parameters_action(self, event_time, node, action, parameters):
        for parameter in parameters:
            if not parameter.name.startswith("qos_overrides"):
                row = [event_time, node, action, parameter.name]
                values = self.get_parameter_value(parameter.value)
                for v in values:
                    row.append(v)
                self.csv_rows.append(row)

    def add_parameters_event(self, event):
        event_time = round(event.stamp.sec + event.stamp.nanosec*1e-9, 2)
        self.add_parameters_action(event_time, event.node, "New", event.new_parameters)
        self.add_parameters_action(event_time, event.node, "Changed", event.changed_parameters)
        self.add_parameters_action(event_time, event.node, "Deleted", event.deleted_parameters)

    def finish(self):
        pprint(self.csv_rows)

if __name__ == "__main__":

    baglocation = sys.argv[1]

    bag_file = BagFileParser.findBagFile(baglocation)
    print("Processing ros2bag" + bag_file)

    parser = BagFileParser(bag_file)
    parameter_events = parser.fetch_parameter_events()
    #pprint(parameter_events)
    print("Found " + str(len(parameter_events)) + " parameter events in bag")

    processor = Parameters()
    processor.prepare();

    for event in parameter_events:
        processor.add_parameters_event(event)

    processor.finish()

clarkebowers, 2021-11-29:
Trying include the launch file again.

<launch>
    <node pkg="turtlesim" exec="turtlesim_node" name="node1">
      <param name="background_r" value="1" />
      <param name="background_g" value="2" />
      <param name="background_b" value="3" />
    </node>
    <node pkg="turtlesim" exec="turtlesim_node" name="node2">
      <param name="background_r" value="4" />
      <param name="background_g" value="5" />
      <param name="background_b" value="6" />
    </node>
    <node pkg="turtlesim" exec="turtlesim_node" name="node3">
      <param name="background_r" value="7" />
      <param name="background_g" value="8" />
      <param name="background_b" value="9" />
    </node>
    <node pkg="turtlesim" exec="turtlesim_node" name="node4">
      <param name="background_r" value="10" />
      <param name="background_g" value="11" />
      <param name="background_b" value="12" />
    </node>
    <node pkg="turtlesim" exec="turtlesim_node" name="node5">
      <param name="background_r" value="13" />
      <param name="background_g" value="14" />
      <param name="background_b" value="15" />
    </node>
</launch>

MichaelOrlov, 2021-12-01:
@clarkebowers What was the DDS QOS settings for publishers and Rosbag2?

clarkebowers, 2021-12-01:
> @clarkebowers What was the DDS QOS settings for publishers and Rosbag2?

This is the provider of DDS
export RMW_IMPLEMENTATION=rmw_cyclonedds_cpp

I have tried a YAML file to force history all. It does not change the result.

/parameter_events:
  history: keep_all

ros2 bag record --all --include-hidden-topics --qos-profile-overrides-path rosbagqos.yaml

1638367867.050641 [0]       ros2: using network interface eth0 (udp/172.27.218.0) selected arbitrarily from: eth0, docker0
[INFO] [1638367867.055968058] [rosbag2_storage]: Opened database 'rosbag2_2021_12_01-14_11_07/rosbag2_2021_12_01-14_11_07_0.db3' for READ_WRITE.
[INFO] [1638367867.056112659] [rosbag2_recorder]: Listening for topics...
[INFO] [1638367867.057017466] [rosbag2_recorder]: Subscribed to topic '/rosout'
[INFO] [1638367867.057144767] [rosbag2_recorder]: Overriding subscription profile for /parameter_events
[INFO] [1638367867.057647770] [rosbag2_recorder]: Subscribed to topic '/parameter_events'

ros2 topic info -v /parameter_events
Type: rcl_interfaces/msg/ParameterEvent

Publisher count: 1

Node name: _ros2cli_daemon_0
Node namespace: /
Topic type: rcl_interfaces/msg/ParameterEvent
Endpoint type: PUBLISHER
GID: a7.6d.10.01.45.3f.df.85.42.97.ed.34.00.00.07.03.00.00.00.00.00.00.00.00
QoS profile:
  Reliability: RELIABLE
  Durability: VOLATILE
  Lifespan: 9223372036854775807 nanoseconds
  Deadline: 9223372036854775807 nanoseconds
  Liveliness: AUTOMATIC
  Liveliness lease duration: 9223372036854775807 nanoseconds

Subscription count: 2

Node name: rosbag2_recorder
Node namespace: /
Topic type: rcl_interfaces/msg/ParameterEvent
Endpoint type: SUBSCRIPTION
GID: 96.fa.10.01.65.68.20.61.ad.d2.ae.9f.00.00.13.04.00.00.00.00.00.00.00.00
QoS profile:
  Reliability: RELIABLE
  Durability: VOLATILE
  Lifespan: 9223372036854775807 nanoseconds
  Deadline: 9223372036854775807 nanoseconds
  Liveliness: AUTOMATIC
  Liveliness lease duration: 9223372036854775807 nanoseconds

Node name: rosbag2_recorder
Node namespace: /
Topic type: rcl_interfaces/msg/ParameterEvent
Endpoint type: SUBSCRIPTION
GID: 96.fa.10.01.65.68.20.61.ad.d2.ae.9f.00.00.15.04.00.00.00.00.00.00.00.00
QoS profile:
  Reliability: RELIABLE
  Durability: VOLATILE
  Lifespan: 9223372036854775807 nanoseconds
  Deadline: 9223372036854775807 nanoseconds
  Liveliness: AUTOMATIC
  Liveliness lease duration: 9223372036854775807 nanoseconds

Everything is reliable with history all.

MichaelOrlov, 2021-12-02:
@clarkebowers I think a key problem here 
> Durability: VOLATILE

It should be `transient-local` to be able to receive messages from history when subscriber joining publisher when he already started publishing messages.

clarkebowers, 2021-12-02:
> @clarkebowers I think a key problem here
> 
> > Durability: VOLATILE
> 
> It should be `transient-local` to be able to receive messages from history when subscriber joining publisher when he has already started publishing messages.

No. I start the capture minutes before the turtle sims are started.

emersonknapp, 2021-12-02:
Just a note to help - you can use triple-backticks ``` to surround formatted blocks of code or output, to make them easier to read.

hidmic, 2021-12-23:
> No. I start the capture minutes before the turtle sims are started.

@clarkebowers that's still not enough. It takes time for the connection between publisher and subscription to establish. For a publisher with a volatile duration policy, all messages sent before that connection settles will be lost (and reliability policies have nothing to do with it). As @MichaelOrlov says, switching to a transient local policy is one possible solution.

BTW, in the general case, there's also the time it takes for `ros2 bag record --all` to [poll](https://github.com/ros2/rosbag2/blob/140169befa6f7179a49a6cfd312f48db47094a23/rosbag2_transport/src/rosbag2_transport/recorder.cpp#L175-L194) the graph for new topics.

hidmic, 2021-12-23:
I'll close this ticket now. Feel free to re-open if you have news.

closed 2021-12-23 by hidmic

ZainSk-Reblade, 2025-07-09:
@clarkebowers So, if we se bag recorder qos durability to Transient Local, then we also have to use same durability for each topic publisher?

MichaelOrlov, 2025-07-09:
@ZainSk-Reblade I would recommend to follow the same approach as we are in the unit tests. i.e., wait until Rosbag2 recorder is ready to receive messages before starting to publish messages from the nodes
https://github.com/ros2/rosbag2/blob/9002ec27b8fbf19042f7282f9173152b670f308e/rosbag2_tests/test/rosbag2_tests/test_rosbag2_record_end_to_end.cpp#L137-L149