ros2/rosbag2 #929: specifying starting time and end time

dongfang-steven-yang, 2021-12-03:
## Description

Hello, 

I just wonder if `ros2 bag play` has functionality like `-s` as in `rosbag play` in ROS 1?
I would like to debug my algorithm at a specific time in a big bag file. For example, the duration of my bag is 10 mins, when I start to replay the topics, I would like to directly jump into 6 mins, so that every time I open a bag, I don't have to wait. 

If there is no such functionality, is there any plan to add it?
Or is there any workaround to solve this issue?

Thanks, 
DF

aprotyas, 2021-12-03:
I don't think the start/stop timestamp feature is exposed in the CLI tool yet, but the infrastructure exists following the work in and https://github.com/ros2/rosbag2/issues/696 and  https://github.com/ros2/rosbag2/issues/821.

aprotyas, 2021-12-03:
Implementation question for maintainers:
For the play start timestamp, can this feature boil down to adding a start timestamp to `rosbag2_py.PlayOptions`, which can then be used to invoke `rosbag2_transport::Player::seek(timepoint)` after construction of a `Player` but before calling `play()` on it? So, right after L115 here.

https://github.com/ros2/rosbag2/blob/2abc50904803a0eae58ee2094542977ec9262db7/rosbag2_py/src/rosbag2_py/_transport.cpp#L109-L128

emersonknapp, 2021-12-03:
These both make sense as additional `PlayOption` fields.

This feature makes sense to me, for starting time, it should be easy to implement by overriding `Player::starting_time_` in `Player::play`. 

The end time could be implemented in the dequeueing mechanism by returning `nullptr` from `Player::peek_next_message_from_queue`, which would stop playback at that time. Alternatively, it could be implemented in the enqueueing mechanism `Player::enqueue_up_to_boundary`, stopping enqueueing when a message past the given timestamp is reached. The result would be approximately the same - maybe slightly better performance to stop enqueueing messages, instead of adding messages to the queue that we don't intend to read.

gbiggs, 2021-12-06:
We have [a PR in our fork[(https://github.com/gbiggs/rosbag2/pull/5) that adds "play until timestamp" functionality. We're planning on upstreaming it later this week or early next week. It might need some revision though.

aprotyas, 2021-12-06:
> We have [a PR in our fork[(https://github.com/[gbiggs/pull/5](https://github.com/gbiggs/rosbag2/pull/5)) that adds "play until timestamp" functionality. We're planning on upstreaming it later this week or early next week. It might need some revision though.

Oops, I didn't realize there was WIP on this. Feel free to close my PR to reduce the noise.

MichaelOrlov, 2023-02-01:
- Closing as done. Since `--start-offset` CLI parameter was added in https://github.com/ros2/rosbag2/pull/931

closed 2023-02-01 by MichaelOrlov

ViktorWalter, 2025-03-19:
This is clearly not completed, as the end time parameter is still missing, and it is crucial for analysis of interesting chunks of larger rosbags.