ros2/rosbag2 #949: Undefined reference to sqlite3_sql and sqlite3_errstr in foxy-future branch

MFernandezCarmona, 2022-01-18:
## Description
Tried to compile foxy-future branch, but something is preventing it from finding sqlite3 libraries. 

## To Reproduce
1. Cloned the branch
```git clone https://github.com/ros2/rosbag2.git -b foxy-future```
2. Installed dependencies
```rosdep install --from-paths src --ignore-src -r -y```
3. Tried to compile
```colcon build --allow-overriding ros2bag rosbag2 rosbag2_compression rosbag2_cpp rosbag2_storage rosbag2_storage_default_plugins rosbag2_transport shared_queues_vendor sqlite3_vendor zstd_vendor```
4. But fails with the following
```Starting >>> sqlite3_vendor
Starting >>> shared_queues_vendor
Starting >>> rosbag2_test_common
Starting >>> zstd_vendor
Finished <<< sqlite3_vendor [0.25s]                                                                                                                                           
Finished <<< rosbag2_test_common [0.28s]
Starting >>> rosbag2_storage
Finished <<< shared_queues_vendor [0.30s]                                                                                     
Finished <<< zstd_vendor [0.31s]
Finished <<< rosbag2_storage [0.26s]                     
Starting >>> rosbag2_storage_default_plugins
--- stderr: rosbag2_storage_default_plugins                              
/usr/bin/ld: librosbag2_storage_default_plugins.so: referencia a `sqlite3_sql' sin definir
/usr/bin/ld: librosbag2_storage_default_plugins.so: referencia a `sqlite3_errstr' sin definir
collect2: error: ld returned 1 exit status
make[2]: *** [CMakeFiles/test_sqlite_wrapper.dir/build.make:167: test_sqlite_wrapper] Error 1
make[1]: *** [CMakeFiles/Makefile2:156: CMakeFiles/test_sqlite_wrapper.dir/all] Error 2
make[1]: *** Se espera a que terminen otras tareas....
/usr/bin/ld: librosbag2_storage_default_plugins.so: referencia a `sqlite3_sql' sin definir
/usr/bin/ld: librosbag2_storage_default_plugins.so: referencia a `sqlite3_errstr' sin definir
collect2: error: ld returned 1 exit status
make[2]: *** [CMakeFiles/test_sqlite_storage.dir/build.make:167: test_sqlite_storage] Error 1
make[1]: *** [CMakeFiles/Makefile2:127: CMakeFiles/test_sqlite_storage.dir/all] Error 2
make: *** [Makefile:141: all] Error 2
---
Failed   <<< rosbag2_storage_default_plugins [0.65s, exited with code 2]

Summary: 5 packages finished [1.33s]
  1 package failed: rosbag2_storage_default_plugins
  1 package had stderr output: rosbag2_storage_default_plugins
  9 packages not processed
```
## Additional info:
Libaries are present, just not found by colcon:
```
locate sqlite3.h
/usr/include/sqlite3.h
/usr/include/spatialite/sqlite3.h

locate sqlite3.a
/usr/lib/x86_64-linux-gnu/libsqlite3.a
```

## System (please complete the following information)
 - OS: Ubuntu 20.04
 - ROS 2 Distro: Foxy
 - Version: release

clalancette, 2022-01-26:
I'm not able to reproduce this error; I can build the `foxy-future` branch just fine here.  Can you run the build again, adding `--event-handlers console_direct+` so we can see more information about what the build is finding?  Also can you give us information about how you are sourcing Foxy?

MFernandezCarmona, 2022-01-27:
Of course:
I have foxy installed from binaries and sourcing it with `source /opt/ros/foxy/setup.bash`
As you can see, I was trying to overlay rosbag, maybe the problem is there. Anyway, here is the output (grab a potato, it's a long one):
```
$ colcon build  --symlink-install --event-handlers console_direct+ --allow-overriding ros2bag rosbag2 rosbag2_compression rosbag2_cpp rosbag2_storage rosbag2_storage_default_plugins rosbag2_test_common rosbag2_transport shared_queues_vendor sqlite3_vendor zstd_vendor
Starting >>> sqlite3_vendor
Starting >>> shared_queues_vendor
Starting >>> rosbag2_test_common
Starting >>> zstd_vendor
-- The C compiler identification is GNU 8.4.0
-- The C compiler identification is GNU 8.4.0
-- The C compiler identification is GNU 8.4.0
-- The C compiler identification is GNU 8.4.0
-- The CXX compiler identification is GNU 8.4.0                                                                                                                   
-- Check for working C compiler: /usr/bin/cc
-- The CXX compiler identification is GNU 8.4.0
-- The CXX compiler identification is GNU 8.4.0
-- Check for working C compiler: /usr/bin/cc
-- Check for working C compiler: /usr/bin/cc
-- The CXX compiler identification is GNU 8.4.0
-- Check for working C compiler: /usr/bin/cc
-- Check for working C compiler: /usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Check for working C compiler: /usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Check for working C compiler: /usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Check for working C compiler: /usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done                                                                                                                           
-- Detecting C compiler ABI info - done
-- Detecting C compiler ABI info - done
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: /usr/bin/c++
-- Detecting C compile features
-- Detecting C compile features - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: /usr/bin/c++
-- Check for working CXX compiler: /usr/bin/c++
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: /usr/bin/c++
-- Check for working CXX compiler: /usr/bin/c++ -- works                                                                                                          
-- Detecting CXX compiler ABI info
-- Check for working CXX compiler: /usr/bin/c++ -- works
-- Check for working CXX compiler: /usr/bin/c++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info
-- Check for working CXX compiler: /usr/bin/c++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compiler ABI info - done                                                                                                                         
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Found ament_cmake: 0.9.9 (/opt/ros/foxy/share/ament_cmake/cmake)
-- Found ament_cmake: 0.9.9 (/opt/ros/foxy/share/ament_cmake/cmake)
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Found ament_cmake: 0.9.9 (/opt/ros/foxy/share/ament_cmake/cmake)
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Found ament_cmake: 0.9.9 (/opt/ros/foxy/share/ament_cmake/cmake)
-- Found PythonInterp: /usr/bin/python3 (found suitable version "3.8.10", minimum required is "3") 
[... 712 lines cut ...]
make[2]: *** [CMakeFiles/test_sqlite_storage.dir/build.make:167: test_sqlite_storage] Error 1
make[1]: *** [CMakeFiles/Makefile2:127: CMakeFiles/test_sqlite_storage.dir/all] Error 2
make: *** [Makefile:141: all] Error 2
--- stderr: rosbag2_storage_default_plugins
/usr/bin/ld: librosbag2_storage_default_plugins.so: undefined reference to `sqlite3_sql'
/usr/bin/ld: librosbag2_storage_default_plugins.so: undefined reference to `sqlite3_errstr'
collect2: error: ld returned 1 exit status
make[2]: *** [CMakeFiles/test_sqlite_wrapper.dir/build.make:167: test_sqlite_wrapper] Error 1
make[1]: *** [CMakeFiles/Makefile2:156: CMakeFiles/test_sqlite_wrapper.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....
/usr/bin/ld: librosbag2_storage_default_plugins.so: undefined reference to `sqlite3_sql'
/usr/bin/ld: librosbag2_storage_default_plugins.so: undefined reference to `sqlite3_errstr'
collect2: error: ld returned 1 exit status
make[2]: *** [CMakeFiles/test_sqlite_storage.dir/build.make:167: test_sqlite_storage] Error 1
make[1]: *** [CMakeFiles/Makefile2:127: CMakeFiles/test_sqlite_storage.dir/all] Error 2
make: *** [Makefile:141: all] Error 2
---
Failed   <<< rosbag2_storage_default_plugins [5.22s, exited with code 2]

Summary: 5 packages finished [12.2s]
  1 package failed: rosbag2_storage_default_plugins
  1 package had stderr output: rosbag2_storage_default_plugins
  9 packages not processed

```

clalancette, 2022-01-27:
I still can't reproduce here.  For reference, here's exactly what I did in a totally fresh Ubuntu 20.04 container:

```
sudo apt update && sudo apt install curl gnupg2 lsb-release
sudo curl -sSL https://raw.githubusercontent.com/ros/rosdistro/master/ros.key  -o /usr/share/keyrings/ros-archive-keyring.gpg
echo "deb [arch=$(dpkg --print-architecture) signed-by=/usr/share/keyrings/ros-archive-keyring.gpg] http://packages.ros.org/ros2/ubuntu $(source /etc/os-release && echo $UBUNTU_CODENAME) main" | sudo tee /etc/apt/sources.list.d/ros2.list > /dev/null
sudo apt update
sudo apt-get install ros-foxy-desktop git python3-colcon-common-extensions ros-foxy-test-msgs pybind11-dev
. /opt/ros/foxy/setup.bash
mkdir -p rosbag2_ws/src
cd rosbag2_ws/src
git clone git@github.com:ros2/rosbag2 -b foxy-future
cd ..
colcon build  --symlink-install --event-handlers console_direct+ --allow-overriding ros2bag rosbag2 rosbag2_compression rosbag2_cpp rosbag2_storage rosbag2_storage_default_plugins rosbag2_test_common rosbag2_transport shared_queues_vendor sqlite3_vendor zstd_vendor
```

I'll suggest two more things:
1. Can you try this in a clean container, following exactly what I did?  That's to ensure that we are on the same page.
2. Can you explain more about your overlay layout?  That can definitely cause problems, which is what may be going on here.

MFernandezCarmona, 2022-01-27:
Hi,
Regarding 1.
I used the docker ros:foxy-ros-base-focal, cloned the repo and I can confirm it works. Compilation was flawless.

Regarding 2.
I tried without any overlay, just sourcing /opt/ros/foxy/setup.bash and trying to compile. Same error. Error is somewhere on my system.

MFernandezCarmona, 2022-01-27:
Ok, I think I found it! I searched for libsqlite3 lib on the docker, and there was only one, here:
```/usr/lib/x86_64-linux-gnu/libsqlite3.so```
But on my machine there were two: that one and "another" here:
```/usr/bin/libsqlite3.so```
No idea about how did I manage to put it there, but as soon as I removed it, everything worked.
We can close the issue here, and I have to find why that library was there...
Thanks

closed 2022-01-27 by MFernandezCarmona