ros2/rosbag2 #955: [foxy-future] building rosbag2 foxy-future branch from source fails because of shared_queues_vendor failing to download the concurrentqueue project

geoporus1, 2022-02-01:
## Description
Building the `rosbag2` `foxy-future` branch fails because of the `shared_queues_vendor` failing to build due to some failed downloads. This could also affect other branches of `rosbag2` at the current time but i am not sure.

### Expected Behavior
`rosbag2` `foxy-future` branch builds successfully.

### Actual Behavior
The build of rosbag2 fails when trying to build the `shared_queues_vendor` package. These are the logs when it fails:
```
--- stderr: shared_queues_vendor
CMake Error at ext-concurrentqueue-stamp/download-ext-concurrentqueue.cmake:159 (message):
  Each download failed!

    
    

make[2]: *** [CMakeFiles/ext-concurrentqueue.dir/build.make:91: concurrentqueue/src/ext-concurrentqueue-stamp/ext-concurrentqueue-download] Error 1
make[1]: *** [CMakeFiles/Makefile2:80: CMakeFiles/ext-concurrentqueue.dir/all] Error 2
make: *** [Makefile:141: all] Error 2
---
Failed   <<< shared_queues_vendor [1min 38s, exited with code 2]
```

## To Reproduce
1. Checkout the `foxy-future` branch of `rosbag2` into your workspace.
2. Proceed to build the workspace.

## System (please complete the following information)
 - OS: The build is done in a docker container with the base image being [osrf/ros:foxy-desktop](https://hub.docker.com/layers/osrf/ros/foxy-desktop/images/sha256-8d5c5e9f91bc9adb3fff08ce542269ea5ebfcb84533ffb189123b1e9921bdf3f?context=explore)
 - ROS 2 Distro: Foxy
 - Version: master

## Additional context
This issue seems to be exactly the same as #611 but that was opened over a year ago, it was concerning a Windows Nightly build and it just seemed to go away as suddenly as it appeared back then? Also in my case, this just randomly happened overnight. One night my containers were building and suddenly all the pipelines are failing because of the `shared_queues_vendor` for `rosbag2` failing to build properly. Noting that if access both the github URLs from which the external projects archives are downloaded in ` rosbag2/shared_queues_vendor/CMakeLists.txt` ([readerwriterqueue](https://github.com/cameron314/readerwriterqueue/archive/ef7dfbf553288064347d51b8ac335f1ca489032a.zip) and [concurrentqueue](https://github.com/cameron314/concurrentqueue/archive/8f65a8734d77c3cc00d74c0532efca872931d3ce.zip)), the downloads work fine and complete without any issues. What would be the best thing to try and fix this?

clalancette, 2022-02-01:
There were a bunch of failures overnight in trying to grab things from GitHub on https://ci.ros2.org.  I can only assume that GitHub had some hiccup overnight, and that caused the problem.  I'm going to close this for now, but if this reproduces again overnight feel free to reopen.

closed 2022-02-01 by clalancette

Henrycious, 2022-02-01:
Had the same error today, but with different packages. 

For me it helped to download the lastest version of cmake from github (https://github.com/Kitware/CMake) (my version now: 3.22.20220101-g0294570). 
After updating Cmake it worked again.
Hope it helps!

geoporus1, 2022-02-01:
@clalancette Actually it is my mistake for not mentioning it in the original description of the issue, but this is happening as we speak. I tried 4 times in a row, just now, to build it and it failed every time with the same error. I believe this is an issue maybe stemming from somewhere else other than a random GitHub hiccup. Can you please re-open this?

@Henrycious thanks for the tip, I will try to do this. Any idea how can I update the CMake from inside a Docker container?

geoporus1, 2022-02-01:
@Henrycious thank you so much for the help! Will post here if this helped! A short check-up showed me that I am indeed using an older version of CMake (3.16.3). Maybe this will really help!