ros2/rosbag2 #956: concurrentqueue md5sum changed

felrock, 2022-02-01:
## Description
concurrentqueue zip file's new md5sum is 64c673dd381b8fae9254053ad7b2be4d, so it breaks the CMakeLists.txt download. It retrys a couple of times before returning a garbage error:
--- stderr: shared_queues_vendor
CMake Error at ext-concurrentqueue-stamp/download-ext-concurrentqueue.cmake:170 (message):
  Each download failed!

### Expected Behavior
md5 to match and build the file.

### Actual Behavior
couple of re-trys and then it returns error

## To Reproduce
** Steps to reproduce the behavior, e.g.
Try to build shared_queue_vendor

## System (please complete the following information)
ubuntu 18.04, master

## Additional context
** Add any other context about the problem here **
I suggest fixing the error message as well, it doesn't print the actual url that can't be downloaded and the problem is clearly the md5sum match but it doesn't mention that in the error message.

Henrycious, 2022-02-01:
You can look there: same error. For me it helped to update to the newest version of CMake

Duplicate of #955

felrock, 2022-02-01:
Yeah, the error roots from the md5sum change in the zip file i posted. Not sure how a update to CMake should do any different here. md5sum exists to prevent changes, and here we have a change. Might even be an attack of some sorts.

geoporus1, 2022-02-01:
@felrock good catch! It is very weird why the md5sum would change out of the blue after so long time. Maybe someone can answer this! Good you opened also an issue on concurrentqueue as well!

clalancette, 2022-02-01:
This still looks like a GitHub backend error, as we saw this error on multiple different packages in multiple different ways overnight.  Right now, I can build `shared_queues_vendor` locally without changes, so I believe that this is fixed on the GitHub backend now.

closed 2022-02-01 by clalancette

felrock, 2022-02-01:
How can a backend issue have an effect on a zip file? Do you know any other repositories that have the same issue? Because I've only seen it with this one

clalancette, 2022-02-01:
> How can a backend issue have an effect on a zip file?

You'll have to ask GitHub support about that.

> Which other repos were effected? I only noticed this one

We had many jobs fail last night with this same error, but on several different packages:

Jobs failing check yaml_cpp_vendor hash:
https://ci.ros2.org/job/nightly_linux_galactic_coverage/286/
https://ci.ros2.org/job/nightly_linux_foxy_coverage/454/
https://ci.ros2.org/view/packaging/job/packaging_linux/2543/
https://ci.ros2.org/view/nightly/job/nightly_linux_release/2178/
https://ci.ros2.org/view/nightly/job/nightly_linux_repeated/2551/

Jobs failing check libyaml_vendor hash:
https://build.ros2.org/view/Fci/job/Fci__benchmark_ubuntu_focal_amd64/272/

Jobs failing shared_queues_vendor hash:
https://ci.ros2.org/view/packaging/job/packaging_linux-rhel/1049/

Jobs failing checking rviz_ogre_vendor hash:
https://build.ros2.org/view/Rci/job/Rci__nightly-release_ubuntu_focal_amd64/588/
https://build.ros2.org/view/Rci/job/Rci__nightly-fastrtps_ubuntu_focal_amd64/587/
https://build.ros2.org/view/Rci/job/Rci__nightly-fastrtps-dynamic_ubuntu_focal_amd64/584/

gavanderhoorn, 2022-02-03:
Related / follow-up on ROS Answers: [Circumvent _vendor projects](https://answers.ros.org/question/395572).

felrock, 2022-02-03:
It's not even close to a follow-up discussion regarding this problem. I refereed to this issue as example as to why _vendor projects_ suck, in my opinion ofc.