ros2/rosbag2 #967: Race condition in discovery of QoS prevents recording topics published with non-default QoS

theseankelly, 2022-03-03:
## Description
There is a race in the DDS participant discovery process such that `ros2 bag record` may discover the presence of a topic based on the existence of a DDS `DataReader` *before* the matching `DataWriter` is discovered.

When this race is lost, `rosbag2_transport` will discover no offered QoS profiles and will instead [subscribe with a default QoS](https://github.com/ros2/rosbag2/blob/master/rosbag2_transport/src/rosbag2_transport/qos.cpp#L110).

In cases where the publisher is providing an incompatible QoS with the default the recording will fail loudly (i.e. if the publisher offers `SensorDataQoS we will see a warning about incompatible reliability) or silently (i.e. if the publisher offers `TRANSIENT_LOCAL` durability, such as previously published `/tf_static` messages will be dropped due to the subscriber's compatible request for `VOLATILE` durability).

Additionally, the `offered_qos_profiles` field in the resulting bag's `metadata.yaml` will be an empty string for the affected topics.

### Expected Behavior
Given a ROS graph with publishers and subscribers using non-default QoS, the expected behavior is to reliably record all the data with `ros2 bag record -a`.

### Actual Behavior
Occasionally (randomly) we miss data from non-default QoS publishers for the duration of a recording. The race in discovery appears to be exacerbated by size of the data and/or network load coming from the publisher.

For example, when using the [ros2_ouster](https://github.com/ros-drivers/ros2_ouster_drivers) driver with 3-4 subscribers, we see the following error 3 or 4 times out of 10 recording sessions:

```
[INFO] [1646323966.741931090] [rosbag2_recorder]: Listening for topics...
[INFO] [1646323966.743906916] [rosbag2_recorder]: Subscribed to topic '/rosout'
[INFO] [1646323966.744695252] [rosbag2_recorder]: Subscribed to topic '/points'
[INFO] [1646323966.746003963] [rosbag2_recorder]: Subscribed to topic '/parameter_events'
[WARN] [1646323966.749143844] [rosbag2_recorder]: New publisher discovered on topic '/points', offering incompatible QoS. No messages will be sent to it. Last incompatible policy: RELIABILITY_QOS_POLICY
[WARN] [1646323966.847185026] [rosbag2_recorder]: A new publisher for subscribed topic /points was found offering RMW_QOS_POLICY_RELIABILITY_BEST_EFFORT, but rosbag already subscribed requesting RMW_QOS_POLICY_RELIABILITY_RELIABLE. Messages from this new publisher will not be recorded.
```

And the metadata for the topic looks like:
```
    - topic_metadata:
        name: /points
        type: sensor_msgs/msg/PointCloud2
        serialization_format: cdr
        offered_qos_profiles: ""
      message_count: 0
```

## To Reproduce
1. Run the [ros2_ouster](https://github.com/ros-drivers/ros2_ouster_drivers) driver
2. Add some network load by launching *4-5* subscribers (`ros2 topic echo /points`)
3. Start a recordings (`ros2 bag record -a`) over and over
4. Repro should hit within 10 iterations of step 3

ALTERNATIVELY, the following steps are not precisely the same repro, but will trigger the same codepath:
1. In terminal 1 create a subscriber by running `ros2 topic echo --qos-reliability best_effort /foo std_msgs/msg/Int32`
2. In terminal 2 run `ros2 record -a`
  a. *NOTE* in the output that the topic `/foo` was added to the bag -- found by the `DataReader`
3. In terminal 3 create a publisher via `ros2 topic pub --qos-reliability best_effort /foo std_msgs/msg/Int32 '{value: 1}'`

The `DataWriter` (and its offered QoS) will be discovered, and terminal 2 (`rosbag record`) will display the warning about incompatible publishers.

## System (please complete the following information)
 - OS: Ubuntu Focal
 - ROS 2 Distro: Rolling
 - Version: 0.13.0

## Additional context
The following node may be used to simulate the Ouster data stream and achieve the repro steps outlined above, though it does depend on a point type defined in the `ros_ouster` package.
```cpp
#include <rclcpp/rclcpp.hpp>
#include <sensor_msgs/msg/point_cloud2.hpp>
#include <pcl_conversions/pcl_conversions.h>

#include <chrono>
#include <cstddef>
#include <memory>
#include <stdexcept>
#include <utility>

class OusterSimNode : public rclcpp::Node
{
public:
  OusterSimNode() : rclcpp::Node("ouster_sim_node")
  {
    // Default to OS0-128, 1024 mode at 10hz
    this->declare_parameter<int>("vres", 128);
    this->declare_parameter<int>("hres", 1024);
    this->declare_parameter<int>("rate_hz", 10);

    int rate_hz;
    this->get_parameter("vres", this->vres_);
    this->get_parameter("hres", this->hres_);
    this->get_parameter("rate_hz", rate_hz);

    if (this->vres_ <= 0)
    {
      throw std::runtime_error("Parameter [vres] must be a positive value.");
    }

    if (this->hres_ <= 0)
    {
      throw std::runtime_error("Parameter [hres] must be a positive value.");
    }

    if (rate_hz <= 0)
    {
      throw std::runtime_error("Parameter [rate_hz] must be a positive value.");
    }

    if (this->vres_ != 32 && this->vres_ != 64 && this->vres_ != 128)
    {
      RCLCPP_WARN(this->get_logger(),
                  "Specified vres [%d] is not a 'real' value (32, 64 or 128)",
                  this->vres_);
    }

    if (this->hres_ != 512 && this->hres_ != 1024 && this->hres_ != 2048)
    {
      RCLCPP_WARN(this->get_logger(),
                  "Specified hres [%d] is not a 'real' value (512, 1024, 2048)",
                  this->hres_);
    }

    if (rate_hz != 10 && rate_hz != 20)
    {
      RCLCPP_WARN(this->get_logger(),
                  "Specified rate_hz [%d] is not a 'real' value (10 or 20)",
                  rate_hz);
    }

    this->pub_ = this->create_publisher<sensor_msgs::msg::PointCloud2>(
        "/points", rclcpp::SensorDataQoS());

    this->timer_ = this->create_wall_timer(
        std::chrono::microseconds(static_cast<int>(1e6 / rate_hz)),
        std::bind(&OusterSimNode::publish_cloud, this));
  }

private:
  void publish_cloud()
  {
    pcl::PointCloud<ouster_ros::Point> cld(this->hres_, this->vres_);
    auto msg = std::make_shared<sensor_msgs::msg::PointCloud2>();
    pcl::toROSMsg(cld, *msg);
    this->pub_->publish(std::move(*msg));
  }

  std::uint32_t vres_;
  std::uint32_t hres_;
  rclcpp::Publisher<sensor_msgs::msg::PointCloud2>::SharedPtr pub_;
  rclcpp::TimerBase::SharedPtr timer_;
};

int main(int argc, char ** argv)
{
  rclcpp::init(argc, argv);
  rclcpp::spin(std::make_shared<OusterSimNode>());
  rclcpp::shutdown();
  return 0;
}
```

theseankelly, 2022-03-03:
Oh, and I forgot to mention that this issue does reproduce on both CycloneDDS and FastDDS -- I believe the race is not necessarily a middlware bug, but is just how the DDS discovery spec works.

MichaelOrlov, 2022-05-10:
Closing this as done in #968 

@theseankelly If you are still interesting to have this fix in Humble release please crate separate PR for backporting #968 and targeting https://github.com/ros2/rosbag2/tree/humble branch.  
Please consider to make it soon we need to finish with it until May19.

closed 2022-05-10 by MichaelOrlov