ros2/rosbag2 #970: Enable discovery after launch (perhaps as an action or service call?)

andrewbest-tri, 2022-03-09:
## Description

** Clear and concise description of the feature that you'd like to exist, from a user's perspective. Don't get into implementation details, describe what you want it to do. Put any consideration of alternate features that might solve your problem as well. **

When running longer experiments, it may be the case that topics start or stop publishing at different times. When collecting a ros bag, it would be extremely helpful for us to be able to trigger discovery on a running bagger so it can pick up new topics. We've tried pre-selecting the topics that may appear. It is prone to human error and may create issues with QOS. We'd like to be able to do this in a programmatic fashion.

## Related Issues

** Is this feature dependent on any other features? Is it part of a larger project? Note here. **

## Completion Criteria

** What needs to be true before we can call this "Done"? Bullet lists are appropriate. **

## Implementation Notes / Suggestions

I suggest using a service call, action call, or possibly adding a flag to the bag verb that would allow a discovery interval. 

We could potentially emit new topics the bagger has found or possibly register it for messages related to new topics (assuming those exist).

** If you have ideas about how this feature might be accomplished, put them here. Note that this is just a suggestion to the implementer, so feel free to speculate. **

## Testing Notes / Suggestions

** All features in this project need tests. Please give some input on cases that will need to be tested - and how the testing might be implemented. **

MichaelOrlov, 2022-03-11:
@andrewbest-tri Starting recording with `ros2 bag record -a` will not solve your problem?   
i.e. running discovery continuously.

andrewbest-tri, 2022-03-15:
I am experimenting again with this. In the past I have had missing data with -a, however I also had significant problems using fastRTPS configs. I may have incorrectly diagnosed the problem!

andrewbest-tri, 2022-03-15:
Ah, yes this is a misdiagnosed problem. I am able to test this with a small dataset. The larger issue relates to https://github.com/ros2/rosbag2/issues/971 and the fact that i have been running without -a so i dont accidentally grab image topics I don't want.

Running with -a works in that I can see it subscribing to new topics (now that I have gotten rid of my fastrtps issues).
Running without -a and a list of topics (some of which are not yet published) also works. It is confusing because once all topics are detected the bagger will print

```
 All requested topics are subscribed. Stopping discovery...
```

which I had been incorrectly understanding as printed when using -a as well. So, this ticket is perhaps a non issue. The real issue will be 971, topic rejection and selection by regex. This would allow me to exclude the image topics I know will inflate my bag size while still catching all other relevant experimental data.

MichaelOrlov, 2022-03-16:
@andrewbest-tri Thnak you for the clarification. I am closing this issue then.

closed 2022-03-16 by MichaelOrlov