ros2/rosbag2 #971: Enable Regex topic selection or rejection

andrewbest-tri, 2022-03-09:
## Description

Deciding which topics to bag could be improved by allowing a regex for topic subscription and rejection. In this way, we could prefix topics we definitely do not want to log, e.g. `ros bag record --exlucude /nolog/*`, or subscribe to entire device feeds without struggling to capture all possible topics, e.g. `ros bag record /imu/*`

This would enable us to expose additional topics from devices and experiment processes without having to modify the bagger constantly. 

## Related Issues

Relates to #970 , the mix of period discovery and regex rejection would be a powerful combination of features.

## Completion Criteria

## Implementation Notes / Suggestions

## Testing Notes / Suggestions

MichaelOrlov, 2022-03-16:
@andrewbest-tri We already have special parameter for excluding topics.
```
            '-x', '--exclude', default='',
            help='Exclude topics containing provided regular expression. '
            'Works on top of --all, --regex, or topics list.')
```
I think it should cover your case.  
Please let us know if not.

andrewbest-tri, 2022-03-16:
So indeed it appears this is available in galactic. We're on foxy but have requisitioned a backport. I will investigate if there are issues using the existing filters. In the future, any advice for how I could find out about these available features before filing issues? Is this documented and I missed it?

emersonknapp, 2022-03-28:
> any advice for how I could find out about these available features before filing issues? Is this documented and I missed it?

Unfortunately, I'd say that documentation is the most lacking part of the project right now.

Significant feature backports can be difficult, while maintaining the required API compatibility, especially if the feature was introduced much later after the code had changed a lot.

andrewbest-tri, 2024-02-26:
@emersonknapp @MichaelOrlov I am closing this for now. There are already issues related to documentation. 

https://github.com/ros2/rosbag2/issues/1412
for example

closed 2024-02-26 by andrewbest-tri