ros2/rosbag2 #973: Nodes stop publishing after ros2 bag record -a

hz658832, 2022-03-14:
## Description
ROS distribution: foxy, built from source on ARM64
ROS nodes: novatel_oem7_driver, velodyne, microstrain_inertial

### Expected Behavior
- topics from novatel GNSS receiver, max. in 10Hz are recorded without lost
- topic /velodyne_points, max. in 10Hz is recorded without lost
- topics /imu/data /mag, in 100Hz are recorded without lost 

### Actual Behavior
- the nodes can be started and the topics can be published on time (used ros2 topic hz ... to check)
- once started ro ros2 bag record -a, node of GNSS receiver stops publishing. No error has been shown
- once stop ros2 bag record -a, the topics of GNSS receiver come back sometime later. 

## System (please complete the following information)
 - OS: Ubuntu 18.04
 - ROS 2 Distro: foxy
 - Version: release

## Additional context
1. is the performance of rosbag2 worse than rosbag1?
2. Was the rosbag2 in original foxy branch patched with the branch foxy-future?

Thanks a lot!
Bests,

Haoming

emersonknapp, 2022-03-28:
To answer the question - the original `foxy` branch was preserved and is the released binary, because `foxy-future` breaks the exposed APIs. The foxy binaries therefore have significantly worse performance than rosbag2 built from `foxy-future` branch.

Have you tried building rosbag2 `foxy-future` from source in your workspace and using that? It will likely have different behavior

hz658832, 2022-03-29:
Hi @emersonknapp,

thanks for the reply!

I have found the reason, it wasn't the rosbag2. 

I am using the novatel_oem7_driver and velodyne, everytime if I have started this two in same time, the novatel_oem7_driver stops publishing message and the device got stucked. I believe this should a bug in the driver from the novatel company. 

Interestingly, same pkgs work fine in ROS1....

But the fact is still there, that rosbag1 shall be faster the rosbag2, is this right?

Thanks again!

Bests,

haoming

closed 2022-03-29 by hz658832