ros2/rosbag2 #999: rosbag2_transport::Recorder .reset()

imstevenpmwyca, 2022-05-03:
## Description
I'm trying to use some tools exposed by the ros2bag_transport package but each time that I try to reset a 
```
auto recorder = std::make_shared<rosbag2_transport::Recorder>(...)
```
I get
```
free(): invalid next size (normal)
```

### Expected Behavior
I believe the code below should just finish normally without an error. Am I doing something wrong?

### Actual Behavior
Instead I'm getting this backtrace
```
(...) received signal SIGABRT, Aborted.
__GI_raise (sig=sig@entry=6) at ../sysdeps/unix/sysv/linux/raise.c:50
50      ../sysdeps/unix/sysv/linux/raise.c: No such file or directory.
(gdb) backtrace
#0  __GI_raise (sig=sig@entry=6) at ../sysdeps/unix/sysv/linux/raise.c:50
#1  0x00007ffff776a859 in __GI_abort () at abort.c:79
#2  0x00007ffff77d529e in __libc_message (action=action@entry=do_abort, 
    fmt=fmt@entry=0x7ffff78ff298 "%s\n") at ../sysdeps/posix/libc_fatal.c:155
#3  0x00007ffff77dd32c in malloc_printerr (
    str=str@entry=0x7ffff79016b8 "free(): invalid next size (normal)")
    at malloc.c:5347
#4  0x00007ffff77dee6c in _int_free (av=0x7ffff7934b80 <main_arena>, 
    p=0x55555557e320, have_lock=<optimized out>) at malloc.c:4322
#5  0x0000555555557b1c in std::__shared_count<(__gnu_cxx::_Lock_policy)2>::~__shared_count (this=<optimized out>, __in_chrg=<optimized out>)
    at /usr/include/c++/9/bits/shared_ptr_base.h:1169
#6  std::__shared_ptr<rosbag2_transport::Recorder, (__gnu_cxx::_Lock_policy)2>::~__shared_ptr (this=<optimized out>, __in_chrg=<optimized out>)
    at /usr/include/c++/9/bits/shared_ptr_base.h:1169
#7  std::__shared_ptr<rosbag2_transport::Recorder, (__gnu_cxx::_Lock_policy)2>::reset (this=0x7fffffff8e50) at /usr/include/c++/9/bits/shared_ptr_base.h:1287
#8  main (argc=<optimized out>, argv=<optimized out>)
```

## To Reproduce
```
#include <rclcpp/rclcpp.hpp>
#include "rosbag2_transport/recorder.hpp"
#include <rosbag2_transport/record_options.hpp>
#include <rosbag2_storage/storage_options.hpp>
#include <rosbag2_cpp/writers/sequential_writer.hpp>

int main(int argc, char** argv) {

  rclcpp::init(argc, argv);

  rosbag2_transport::RecordOptions record_options{false,false,{"/topic"},rmw_get_serialization_format(),std::chrono::milliseconds(100)};
  rosbag2_storage::StorageOptions storage_options{"filename","sqlite3"};

  auto writer_=std::make_shared<rosbag2_cpp::Writer>();
  auto recorder = std::make_shared<rosbag2_transport::Recorder>(std::move(writer_),storage_options,record_options,"record_node");

// The 'record_node' exists
  rclcpp::sleep_for(std::chrono::seconds(5));

// Crashes with the error mentioned
  recorder.reset();

  rclcpp::shutdown();

  return 0;
}
```

## System
 - OS: Ubuntu 20.04.4 LTS
 - ROS 2 Distro: Galactic
 - Version: Latest

MichaelOrlov, 2022-05-06:
@imstevenpmwyca I confirm this problem. The issue that that we currently don't have **stop** operation in `Recorder` and destructor doesn't fully care about appropriate interruption and finish of the recorder.  

This issue will be fixed soon when we `ApexAI` will upstream `stop` command for the Recorder.

For a while I can suggest a workaround by calling 
```cpp
rclcpp::shutdown();
```
before resetting recorder.  
If it will not help in your case as expected, please try to put some delay after `rclcpp::shutdown();`

MichaelOrlov, 2024-05-05:
- Closing as fixed in the #1300

closed 2024-05-05 by MichaelOrlov