ros2/rosbag2 #1000: ros2 bag can't be shut down with ctrl + c when the bag is paused

Kaju-Bubanja, 2022-05-04:
## Description
ros2 bag can't be shut down with ctrl + c when the bag is paused, but it can also not be resumed after sending ctrl + c once. It enters a frozen state where only ctrl + z helps.

### Expected Behavior
Clean shutdown like when the bag is not paused

### Actual Behavior
The program freezes

## To Reproduce
1. Play a bag
2. Pause bag
3. Press ctrl + c
4. Observe that it doesn't shut down
5. Press ctrl + z
6. Observe that it shuts down

## System (please complete the following information)
 - OS: Ubuntu 22.04
 - ROS 2 Distro: Ros rolling
 - Version: master(built from source)

MichaelOrlov, 2022-05-04:
@Kaju-Bubanja I confirm this issue. I came across on it recently too.  
Thinking how to better fix it.

Kaju-Bubanja, 2022-05-05:
@MichaelOrlov While I have your attention :P May I ask you to have a look at this issue: https://github.com/ros2/rosbag2/issues/966 ? The ctrl + c or z issue is a minor cosmetic thing, but the issue 966 is really a blocker for me right now, because I can't play back a large bag file which is split. The issue seems orphaned but easy to fix(I assume, could always turn out very hard ofc), if you know more about the rosbag2 internals.

MichaelOrlov, 2022-05-06:
@Kaju-Bubanja After some analysis I can say that fix for this particular issue is going to be in one line and most probably will need to add one more regression test.
Although for #966 it is quite more complicated and entire feature for splitting bags need more "love".  
We have multiple issues with splitting bags https://github.com/ros2/rosbag2/pull/936, https://github.com/ros2/rosbag2/pull/866 and https://github.com/ros2/rosbag2/issues/647. One of them is the high probability to lose messages in some cases.
Please refer to the https://github.com/ros2/rosbag2/issues/647 for details.

Giving that I wouldn't recommend to avoid using splitting bag feature for a while.

Kaju-Bubanja, 2022-05-06:
I see, unfortunate because I have a weeks worth of data which I can't analyse. I guess I will dig into the code to see what I can do.
I guess wouldn't is a typo and should be would?

MichaelOrlov, 2022-05-06:
You can try to put correct start time in `metadata.yaml` file manually.  Just need to figure out the correct one.
To do this you can open up `.db3` file in some `SQLite3` db manager and see timestamp in first message.

Kaju-Bubanja, 2022-05-06:
The thing is the times are correct everywhere, that's what confused me. They are correct in the metadata.yaml and in the individual bags. I think the reader is somehow broken.

audrow, 2022-05-06:
Hey @MichaelOrlov, for the Humble release, do you think we should say in the release documentation that splitting bags is an experimental feature?

MichaelOrlov, 2022-05-06:
@audrow Not sure that it's a good wording to say that it's experimental since it's already exists for a while, more than year or so.  
May be better way to refer to the known #647 and #966 issues?

Kaju-Bubanja, 2022-05-07:
Maybe the best would be to have a warning where the doc is on how to use it. Because for me it was quite a surprise, the rest of rosbag2 works good, so i expected the split feature to work too and didnt test playback first before collecting data. This just in case the humble release happens before the fix. Better be warned then surprised later.

audrow, 2022-05-09:
> Maybe the best would be to have a warning where the doc is on how to use it.

Perhaps a warning would be a good idea.

> Not sure that it's a good wording to say that it's experimental since it's already exists for a while, more than year or so.

The nice thing about calling it "experimental" is that, IIUC, it doesn't lock in the API/ABI, so that they can change if needed to make fixes. To me it basically means that it's still evolving, which doesn't necessarily relate to how long the feature has been around.

MichaelOrlov, 2022-05-10:
@Kaju-Bubanja I've prepared fix https://github.com/ros2/rosbag2/pull/1002 for this bug.

MichaelOrlov, 2022-05-11:
@audrow 
> The nice thing about calling it "experimental" is that, IIUC, it doesn't lock in the API/ABI, so that they can change if needed to make fixes. To me it basically means that it's still evolving, which doesn't necessarily relate to how long the feature has been around.

Ok. I agree.
Wouldn't you mind to put such warning in release doc by yourself?

audrow, 2022-05-12:
> Wouldn't you mind to put such warning in release doc by yourself?

Will do.

closed 2022-05-14 by MichaelOrlov

audrow, 2022-05-23:
FYI, here's the release note in the known issues section: https://github.com/ros2/ros2_documentation/pull/2643.

MichaelOrlov, 2022-05-23:
Thank you @audrow. Much appreciated.

PR #1002 (2022-05-14): Fix for rosbag2::Player freeze when ctrl+c in pause mode
Fix for rosbag2::Player freeze when ctrl+c in pause mode
Addressing bug report #1000

- Closes #1000