ros2/rosbag2 #1011: Warnings about 'Topic ... has unknown type' spam when specifying topics to be recorded which do not exist

wjwwood, 2022-05-11:
## Description

When asking rosbag to record a list of topics when some of those topics do not exist _and_ some of those topics have unknown types results in continuous spam of messages like this:

```
[WARN] [1652302035.004255795] [ROSBAG2_TRANSPORT]: Topic '/some/topic' has unknown type 'msg_pkg/msg/SomeMessage' . Only topics with known type are supported. Reason: 'package 'msg_pkg' not found, searching: [/some/workspace/path/some_package, ...]
```

These messages are printed over and over again, despite there being some logic to try and only print it once per type:

https://github.com/ros2/rosbag2/blob/a076560b09a6d80ce91815f4a03e113a0331a68d/rosbag2_transport/src/rosbag2_transport/topic_filter.cpp#L154-L170

### Expected Behavior

This kind of error message is printed only once and does not spam the console while recording in this particular kind of scenario.

### Actual Behavior

The error message is printed repeatedly.

## To Reproduce

1. Run a publisher using a type not available to rosbag (either in another workspace or on another machine)
2. Ask `ros2 bag record` to record that topic and a topic that doesn't exist
3. Observe the error about the topic using a type unknown to rosbag is printed over and over

## System (please complete the following information)

 - OS: Ubuntu 20.04
 - ROS 2 Distro: Rolling
 - Version: `0.14.1`

## Additional context

I believe the issue is this logic for when multiple topics are requested but not immediately found:

https://github.com/ros2/rosbag2/blob/a060c1da8fac91546c77c223cbabcaef85902b01/rosbag2_transport/src/rosbag2_transport/recorder.cpp#L175-L194

This causes this function to be called in a loop:

https://github.com/ros2/rosbag2/blob/a060c1da8fac91546c77c223cbabcaef85902b01/rosbag2_transport/src/rosbag2_transport/recorder.cpp#L196-L202

Which as you can see instantiates a new `TopicFilter` every time, and the `TopicFilter` holds the state that checks if the warning should be printed again:

https://github.com/ros2/rosbag2/blob/a076560b09a6d80ce91815f4a03e113a0331a68d/rosbag2_transport/src/rosbag2_transport/topic_filter.cpp#L160-L161

So the likely solution is to find a way to extend the lifetime of the `TopicFilter` instance.

closed 2022-05-25 by ihasdapie

Jannkar, 2023-09-11:
Hey! Was this MR back-ported to Humble? I'm currently running rosbag2 version 0.15.7 in Humble, and I'm getting the console full of warning spam. This happens for topics that I'm not even recording, so could this be a combination of this spamming issue and #1444 ?

MichaelOrlov, 2023-09-12:
Hi @Jannkar No. The fix hasn't yet been backported since backporting is not straightforward because it involves ABI changes.   
See details in the https://github.com/ros2/rosbag2/pull/1434#issuecomment-1671665534

Jannkar, 2023-09-12:
Thanks for letting me know!

commit afb33210 (2022-05-22): fix #1011: persist TopicFilter to avoid warning spam
PR #1018 (2022-05-25): Reduce message spam when topics to be recorded do not exist
Commits extend lifetime of `TopicFilter` to avoid emitting warning messages on instantiation. `topic_filter.hpp` moved to make it accessible from `recorder.hpp`.

- closes #1011
commit d97e2911 (2022-05-25): Reduce message spam when topics to be recorded do not exist (#1018)

* move topic_filter.hpp to expose to recorder.hpp

* fix #1011: persist TopicFilter to avoid warning spam

* address pr comments