ros2/rosbag2 #1015: :man_farmer: `Killed signal terminated program cc1plus` while building `rosbag2_transport/test_play_loop.cpp.o`

Blast545, 2022-05-13:
## Description
I don't think it's coincidence two coverage jobs failed with:

```
00:54:42.817 [ 43%] Building CXX object CMakeFiles/test_play_services__rmw_cyclonedds_cpp.dir/test/rosbag2_transport/test_play_services.cpp.o
00:54:42.817 [ 44%] Building CXX object CMakeFiles/test_play_publish_clock__rmw_cyclonedds_cpp.dir/test/rosbag2_transport/test_play_publish_clock.cpp.o
00:54:42.817 [ 44%] Building CXX object CMakeFiles/test_play_loop__rmw_cyclonedds_cpp.dir/test/rosbag2_transport/test_play_loop.cpp.o
00:54:42.817 [Kc[Kfatal erro[KKilled signal terminated program cc1plus
00:54:42.817 compilation terminated.
00:54:42.817 gmake[2]: *** [CMakeFiles/test_play_seek__rmw_cyclonedds_cpp.dir/build.make:76: CMakeFiles/test_play_seek__rmw_cyclonedds_cpp.dir/test/rosbag2_transport/test_play_seek.cpp.o] Error 1
00:54:42.817 gmake[1]: *** [CMakeFiles/Makefile2:966: CMakeFiles/test_play_seek__rmw_cyclonedds_cpp.dir/all] Error 2
00:54:42.817 gmake[1]: *** Waiting for unfinished jobs....
00:54:42.817 [ 45%] Linking CXX executable test_play_publish_clock__rmw_cyclonedds_cpp
00:54:42.817 [ 46%] Linking CXX executable test_play_loop__rmw_cyclonedds_cpp
00:54:42.817 [ 46%] Built target test_play_publish_clock__rmw_cyclonedds_cpp
00:54:42.817 [ 46%] Built target test_play_loop__rmw_cyclonedds_cpp
00:54:42.817 [ 46%] Linking CXX executable test_play_services__rmw_cyclonedds_cpp
00:54:42.817 [ 46%] Built target test_play_services__rmw_cyclonedds_cpp
00:54:42.817 gmake: *** [Makefile:146: all] Error 2
00:54:42.817 ---
```

Failed coverage builds:
https://ci.ros2.org/job/nightly_linux_coverage/1606/
https://ci.ros2.org/job/nightly_linux_humble_coverage/17/

This issue is mainly to keep track, may happen again over the weekend. 

### Expected Behavior
Package gets built without failures.

### Actual Behavior
Fails to build.

## To Reproduce
Run a buildfarm coverage job.

## System (please complete the following information)
 - OS: Linux Jammy
 - ROS 2 Distro: Humble and Rolling
 - Version: `main`

Blast545, 2022-06-01:
Potentially related to: https://github.com/ros2/rosbag2/issues/872

clalancette, 2022-06-09:
Yes, this is almost certainly the same thing as #872.  Since that one has some suggestions for improvement, I'm going to actually copy a few new tidbits out of this one into #872, and close it in favor of that one.