ros2/rosbag2 #1019: ros2 bag play error on windows foxy - [rosbag2_transport]: Failed to play: cannot get file size

karthiknit1, 2022-05-23:
## Description
when there is only rosbag2_2021_09_27-09_11_59.db3 file and no metadata.yaml file, when I tried to play the bag file using the command "ros2 bag play rosbag2_2021_09_27-09_11_59.db3", the following error is encountered in ROS 2 Foxy terminal on windows.

[ERROR] [1652675142.815790900] [rosbag2_transport]: Failed to play: cannot get file size: The file pointer cannot be set on the specified device or file.

### Expected Behavior
Bag file should be played successfully.

### Actual Behavior
ros2 bag play rosbag2_2021_09_27-09_11_59.db3
ros2 bag info rosbag2_2021_09_27-09_11_59.db3
Both the above commands are leading to the below error on windows only:
[ERROR] [1652675142.815790900] [rosbag2_transport]: Failed to play: cannot get file size: The file pointer cannot be set on the specified device or file.

## To Reproduce
** Steps to reproduce the behavior, e.g.
1. record a bag file.(around 8GB) in ROS 2 Foxy terminal.
2. play the bag using "ros2 bag play bagFileName". Here instead of bag folder, provide bag file name(.db3)

## System (please complete the following information)
 - OS: Windows
 - ROS 2 Distro: Foxy
 - Version: Installed binary of latest Foxy
 
## Additional context
** Add any other context about the problem here **

karthiknit1, 2022-11-09:
Any update on this issue? This seems to be most recurring problem on win64 for large ros2 bags.

MichaelOrlov, 2022-11-09:
What behavior if specify folder with .db3 file and `metadata.yaml` file? for `ros2 bag play` command? 
For the same recorded .db3 file.
Also does this issue still reproduces for ROS2 `Rolling` distro?

karthiknit1, 2022-11-28:
Hello Michael,

If I specify folder, I do not see any issue. But this issue with file path instead of folder path exists in ros2 Humble distro as well (issue seen in windows). I didn't get chance to check it on Rolling distro.

The reason for providing file path is to read the splitted bag files (if any) recorded in a bag folder.

Thanks,
Karthik Reddy

MichaelOrlov, 2022-11-30:
Preliminary analysis:
Exception with rrror message "cannot get file size: The file pointer cannot be set on the specified device or file." coming from 
https://github.com/ros2/rosbag2/blob/197b09481a167d7fe9414c23945a778288f44a62/rosbag2_storage_default_plugins/src/rosbag2_storage_default_plugins/sqlite/sqlite_storage.cpp#L204-L206
Which is called from
https://github.com/ros2/rosbag2/blob/197b09481a167d7fe9414c23945a778288f44a62/rosbag2_storage_default_plugins/src/rosbag2_storage_default_plugins/sqlite/sqlite_storage.cpp#L357
`get_relative_file_path()` pointing to the `relative_path_` which is assigned to the `uri` during calling `open(..)` method.
And I am not mistaken this uri is the same as provided in CLI.

MichaelOrlov, 2022-11-30:
It seems `relative_path_` should have a valid value since we checking it for existence and even opening database file in `open(..)` method
https://github.com/ros2/rosbag2/blob/197b09481a167d7fe9414c23945a778288f44a62/rosbag2_storage_default_plugins/src/rosbag2_storage_default_plugins/sqlite/sqlite_storage.cpp#L89-L99
We calling `storage_->get_metadata();` when storage already has been opened in fabric method 
https://github.com/ros2/rosbag2/blob/197b09481a167d7fe9414c23945a778288f44a62/rosbag2_cpp/src/rosbag2_cpp/readers/sequential_reader.cpp#L104-L110

Conclusion: It looks like problem in underlying implementation of the `rcpputils::fs::path::file_size()` from `ros2/rcpputils/src/filesystem_helper.cpp` in particularly for Windows platform.
Need to check some relevant issues for this particular function.

MichaelOrlov, 2023-02-01:
Closing this issue since the are nothing to fix in rosbag2. This is underlying issue in `rcpputils` package.  
Feel free to re-open if disagree.

closed 2023-02-01 by MichaelOrlov