ros2/rosbag2 #1020: Determining the message type in a compression plugin.

benbarron, 2022-05-24:
I have a use case where I would like to determine the type of a message in my compression plugin. This could be accomplished by adding the type_name to `SerializedBagMessage`.

The new data structure. 
```cpp
struct SerializedBagMessage
{
  std::shared_ptr<rcutils_uint8_array_t> serialized_data;
  rcutils_time_point_value_t time_stamp;
  std::string topic_name;
  std::string type_name;
};
```

 Is this something that is doable? Is there another way to determine the type? I saw a similar issue https://github.com/ros2/rosbag2/issues/677 but it looked like it has been inactive for about a year now. Please let me know what would be needed to get this done. I am willing to contribute if needed.

MichaelOrlov, 2022-05-25:
@benbarron It's sort of implemented in new mcap storage plugin please refer to the https://github.com/ros-tooling/rosbag2_storage_mcap/blob/main/rosbag2_storage_mcap/src/mcap_storage.cpp#L357-L371

MichaelOrlov, 2024-05-05:
- Closing this request in favor of the other opened #1553. 
After #1553 is implemented, one can match topic_id to the topic_type stored in metadata.
We have complaints from other users about the excessive memory usage of the rosbag2 recorder by storing the topic name as a string in the `struct SerializedBagMessage`.  Adding one more string field, aka `type_name`, will only worsen the problem with excessive memory usage,

closed 2024-05-05 by MichaelOrlov