ros2/rosbag2 #1025: Extra bytes in serialized ros bag data

QwertyBobberson, 2022-06-13:
## Description
I was looking at serialized bag data that came from ros2 and I noticed that there are occasionally 4 extra zeroed out bytes being added where they do not belong. This data is supposed to be in the control_msgs/JointTrajectoryControllerState format, and the extra 4 zero bytes are occasionally being inserted between the length of a double array and the actual data.

### Expected Behavior
The arrays should only be serialized as the length and the data.

### Actual Behavior
There are 4 completely zeroed out bytes sometimes appearing at random between the length and data

## System
 - OS: The data was recorded on Ubuntu and read from in windows
 - ROS 2 Distro: Rolling

## Additional context
The extra data either appears in multiple arrays in one frame or 0

MichaelOrlov, 2022-06-16:
@QwertyBobberson Could you please add more details how we can reproduce this issue or how to see those 4 extra zeroed bytes?

QwertyBobberson, 2022-06-19:
@MichaelOrlov I don't know how the recording was done, it was given to me by someone else, so I don't know how to recreate the issue. I viewed the data by reading the bag file with the sqlite3 api for c++ and printing the data field (a BLOB) after storing it in a char*.

MichaelOrlov, 2022-07-26:
@QwertyBobberson Extra bytes could be due to memory alignment used by compiler.   
sizeof(some_struct) not always return what you would expect.  
Can you double check that `sizeof` of the structure with extra bytes return value as you are expecting on the same build?

QwertyBobberson, 2022-09-13:
Sorry, I got reassigned to a different project and forgot about this. I no longer work there at all, so I'm going to close this issue.

To answer your question though, I did check this when I first encountered the issue, the sizeof returned the values I expected.

closed 2022-09-13 by QwertyBobberson