ros2/rosbag2 #1026: Support topic remapping in `ros2 bag convert` YAML syntax

asymingt, 2022-06-20:
## Description

It would be useful to be able to specify topic remaps in the ros2 bag converter YAML syntax. So, for example if we recorded some topics `/input` and `/output` in our experimental run, when we `ros2 bag convert ...` the bag we can remap `/output -> /original/output`. In replay we might have some algorithm that consumes `/input` to produce a new `/output`, which we could then compare to `/original/output` in a metrics pipeline. I realize we can do this in a launch context when `ros2 bag play ...` is called. However, this mixes data preparation steps with launch files.

## Related Issues

No. But this is connected more broadly to bag patching, which I'll post separately.

## Completion Criteria

Supporting YAML syntax for declaring topic remaps to be done *before* the topic filter is applied when converting a bag using the `ros2 bag convert` tool, and specifying a output filter YAML. The remaps should work exactly like `--remap` in ros2 play and hopefully support wildcards. 

## Implementation Notes / Suggestions

The YAML syntax for an output filter might support the `remap` keyword. The example below is equivalent for `ros2 bag play --remap /output:=/original/output`  into a new bag, and then running the converter. Now the converter does this for you.

```yaml
output_bags:
- uri: /tmp/rgsw.bag
  storage_id: sqlite3
  remap:
    - "/output:=/original/output"
  topics: ["/input", "/original/output"]
```

## Testing Notes / Suggestions

It should be trivial to setup an example where some arbitrary test message is remapped from some old to some new topic, and verify that the converted bag contains the original message on the new topic. A more advanced test might also include wildcard mapping.

MichaelOrlov, 2022-06-22:
@asymingt Thank your for new feature request.
As usual PRs are welcome for consideration.
Form my side as a help with design and implementation I created draft https://github.com/ros2/rosbag2/pull/1028

asymingt, 2022-06-22:
@MichaelOrlov can you maybe take a look at my PR (https://github.com/ros2/rosbag2/pull/1030)  to try and help me understand how I'm using `rcl_remap_topic_name` incorrectly?