ros2/rosbag2 #1035: rosbag2 compilation stuck in zstd_vendor node does not move

yunerJin, 2022-06-28:
## Description
** When I compile rosbag2, the progress compilation is stuck in the zstd_vendor node **

[image]

How to solve this problem!

## System (please complete the following information)
 - OS: Docker(Ubuntu 20.04)
 - ROS 2 Distro: Foxy
 - Version: foxy branch

MichaelOrlov, 2022-07-26:
@yunerJin Do you have access to the internet on machine where you are running build? 
It might be intermittent problem with access to the zstd third party dependencies when trying to pull it out via internet during the build.

MichaelOrlov, 2023-02-01:
Closing as stale and not reproducible issue.

closed 2023-02-01 by MichaelOrlov