ros2/rosbag2 #1041: Play back of split bagfile does not respect proper rates.

PeterQFR, 2022-07-13:
## Description
When playing back a bagfile that has been split with -d 30, the data is published as fast as it could be read. This means that a 200s bagfile is played back in 1-2 seconds.

### Expected Behavior
As per play of single bagfile, we expect that data is played back at the same rate as it was recorded.

### Actual Behavior
ros2 bag play <bagname> loads all split bagfiles in database and immediately publishes recorded messages as fast as they can be read from the data base. This means that a 200s bagfile can play in 1-2 secs. 
I can see the messages all being replayed in ultra fast time. Using ros2 topic hz, it measures a 100Hz topic at 4000 - 750 Hz.

## To Reproduce
** Steps to reproduce the behavior, e.g.
record a bagfile 
ros2 bag record -d 30 -o /data/bags/$(var mission_prefix) --max-cache-size 2000000000 <topic1> <topic2>
record for 1-2 mins. 

play back bagile
ros2 bag play /data/bags/$(var mission_prefix).

if I set a low rate using -r 0.1 it appears to only replay the later bags.
## System (please complete the following information)
 - OS: Ubuntu 20.04
 - ROS 2 Distro: Galactic
 - Version: apt package manager release

## Additional context
** Add any other context about the problem here **

MichaelOrlov, 2022-07-14:
@PeterQFR This issue could be related to the https://github.com/ros2/rosbag2/issues/966, please double check if it's has been fixed on latest master.

PeterQFR, 2022-07-15:
I did an update from apt yesterday, should that fix have been included there? If so it is still an issue. Or do I need to build from source?

MichaelOrlov, 2022-07-15:
Unfortunately will need to build from source.

MichaelOrlov, 2023-02-01:
Closing as stale and most probably a duplicate of already fixed #966 issue.  
Feel free to reopen if issue still reproducible.

closed 2023-02-01 by MichaelOrlov