ros2/rosbag2 #1042: Tool to create a video from images recorded in a rosbag

ivanpauno, 2022-07-15:
## Description

I have a script to create a video from a images recorded in a rosbag.
I was wondering what's the correct place to add it, and I thought the best place would be the `rosbag2` repo.

Should that be part of `ros2bag`?
Should that be a separate script?
Should that be contributed like an example of how to use `rosbag2_py` API?

Any of the alternatives sound good to me.
I can also add it in another repo if it doesn't sound like a good idea to add it here.

## Related Issues

## Completion Criteria

* Decide where the code should be added.
* Open PR adding the code.
  * Initial implementation is in https://gist.github.com/ivanpauno/d97acc45675acf07a3ad84ee99036608, but it really needs to be polished.

## Implementation Notes / Suggestions

## Testing Notes / Suggestions

jacobperron, 2022-07-15:
We should also consider the dependencies that the example brings, notably OpenCV and `cv_bridge`. It may not be desirable to have the `rosbag2` repository depend on these wrt making releases.

clalancette, 2022-07-15:
> notably OpenCV and `cv_bridge`

OpenCV isn't a big deal; we already depend on it in https://github.com/ros2/demos/blob/rolling/image_tools/package.xml .  cv_bridge would be a new dependency, and we'd seriously have to think about whether we want to make that a core dependency.  I'd really prefer not to, but if it is really important we can consider it.

ivanpauno, 2022-07-15:
Alternatively, we could create a repo in a ros github organization (`ros2` or `ros-tooling`) sound appropriate, and release that package independently.
I think we also mentioned `demos` or `examples` as a place to add the new package, but they don't seem like a great place for the tool and we would have the same dependencies issues.

> notably OpenCV and cv_bridge

I think I can get rid of both and use something like ffmpeg API directly.
I will investigate a bit and try to do that.

@jacobperron to iterate on the script, I will create a repo.
I can later transfer it to another org or create a PR adding the package to an existing repo, whichever we decide is more appropriate.

ivanpauno, 2022-07-15:
I think it would be could to add the script as a `ros2bag` verb, e.g. `ros2 bag to_video ...`.
AFAIS, we can do that from another package (as we're using entrypoints to discover verbs), so it doesn't matter if we add the code here or in another place.

Though if we get rid of the extra dependencies, I think it makes sense to add it here.

ivanpauno, 2022-07-20:
> I think I can get rid of both and use something like ffmpeg API directly.

I took a look to that.
The alternatives are `ffmpeg-python` or `pyav`.
The first isn't released as a debian package and can only be installed through pip, so I discarded that one.

For pyav, converting from `sensor_msgs/Image` to their frame type requires a lot of care for the different image encodings, which is code similar in complexity to cv_bridge, so I don't think it's worth following that path.

---

I have created a repo with the tool [here](https://github.com/ivanpauno/rosbag2_to_video), and added a `ros2 bag to_video` verb there.