ros2/rosbag2 #1047: Error During Colcon Build

SimeonOA, 2022-07-20:
## Description
I am installing ROS2 from source following the instructions here: [from Source](https://docs.ros.org/en/rolling/Installation/Alternatives/Ubuntu-Development-Setup.html) and [maintaining source checkout](https://docs.ros.org/en/rolling/Installation/Maintaining-a-Source-Checkout.html) in a Docker container on Jetson Nano. 

### Expected Behavior
I am building ROS 2 from source in a Docker container. I expect it to complete build without error.

### Actual Behavior
`colcon build` currently fails on rosbag2_storage with the error shown below
```
--- stderr: rosbag2_storage
In file included from /opt/ros/rolling/src/ros2/rosbag2/rosbag2_storage/src/rosbag2_storage/metadata_io.cpp:26:0:
/opt/ros/rolling/src/ros2/rosbag2/rosbag2_storage/include/rosbag2_storage/yaml.hpp: In function ‘T YAML::decode_for_version(const YAML::Node&, int)’:
/opt/ros/rolling/src/ros2/rosbag2/rosbag2_storage/include/rosbag2_storage/yaml.hpp:82:38: error: ‘const class YAML::Node’ has no member named ‘Mark’
     throw TypedBadConversion<T>(node.Mark());
                                      ^~~~
/opt/ros/rolling/src/ros2/rosbag2/rosbag2_storage/include/rosbag2_storage/yaml.hpp:88:36: error: ‘const class YAML::Node’ has no member named ‘Mark’
   throw TypedBadConversion<T>(node.Mark());
                                    ^~~~
/opt/ros/rolling/src/ros2/rosbag2/rosbag2_storage/include/rosbag2_storage/yaml.hpp: In function ‘T YAML::decode_for_version(const YAML::Node&, int) [with T = std::vector<rosbag2_storage::TopicInformation>]’:
/opt/ros/rolling/src/ros2/rosbag2/rosbag2_storage/include/rosbag2_storage/yaml.hpp:89:1: warning: control reaches end of non-void function [-Wreturn-type]
 }
 ^
/opt/ros/rolling/src/ros2/rosbag2/rosbag2_storage/include/rosbag2_storage/yaml.hpp: In function ‘T YAML::decode_for_version(const YAML::Node&, int) [with T = rosbag2_storage::TopicInformation]’:
/opt/ros/rolling/src/ros2/rosbag2/rosbag2_storage/include/rosbag2_storage/yaml.hpp:89:1: warning: control reaches end of non-void function [-Wreturn-type]
 }
 ^
/opt/ros/rolling/src/ros2/rosbag2/rosbag2_storage/include/rosbag2_storage/yaml.hpp: In function ‘T YAML::decode_for_version(const YAML::Node&, int) [with T = rosbag2_storage::TopicMetadata]’:
/opt/ros/rolling/src/ros2/rosbag2/rosbag2_storage/include/rosbag2_storage/yaml.hpp:89:1: warning: control reaches end of non-void function [-Wreturn-type]
 }
 ^
make[2]: *** [CMakeFiles/rosbag2_storage.dir/src/rosbag2_storage/metadata_io.cpp.o] Error 1
make[1]: *** [CMakeFiles/rosbag2_storage.dir/all] Error 2
make: *** [all] Error 2
---
Failed   <<< rosbag2_storage [47.5s, exited with code 2]

```

## To Reproduce
Build ROS 2 from source following the instructions [here](https://docs.ros.org/en/rolling/Installation/Alternatives/Ubuntu-Development-Setup.html) and [here](https://docs.ros.org/en/rolling/Installation/Maintaining-a-Source-Checkout.html) 

## System (please complete the following information)
 - OS:  Ubuntu 20.04 
 - ROS 2 Distro: Rolling
 -  Installation type:  Source

## Additional context
This is the relevant section of code from the Dockerfile:

```
RUN mkdir -p ${ROS_ROOT}/src && \
    cd ${ROS_ROOT} && \
    wget https://raw.githubusercontent.com/ros2/ros2/master/ros2.repos && \
    vcs import src < ros2.repos && \
    vcs pull src && \

    # Orocos-kdl
    git clone https://github.com/orocos/orocos_kinematics_dynamics && \
    cd ${ROS_ROOT}/orocos_kinematics_dynamics && \
    git submodule update --init && \
    rm -rf ${ROS_ROOT}/src/ros2/rclcpp

COPY ./rclcpp ${ROS_ROOT}/src/ros2/

    # install dependencies using rosdep
RUN apt-get update && \
    cd ${ROS_ROOT} && \
    rosdep init && \
    rosdep update && \
    rosdep install -y \
          --ignore-src \
       --from-paths src \
	  --skip-keys "fastcdr rti-connext-dds-6.0.1 urdfdom_headers ignition-cmake2 ignition-math6 python3-pykdl" && \
    rm -rf /var/lib/apt/lists/* && \
    apt-get clean && \
    # build it!
    colcon build \
    	--symlink-install \
        --merge-install \
        --cmake-args -DCMAKE_BUILD_TYPE=Release && \
    # remove build files
    rm -rf ${ROS_ROOT}/src && \
    rm -rf ${ROS_ROOT}/logs && \
    rm -rf ${ROS_ROOT}/build && \
```

MichaelOrlov, 2022-07-26:
@SimeonOA It looks like this error relates to the using old version of the `yaml_cpp` library.
See https://stackoverflow.com/questions/25962473/does-yamlnode-have-a-mark-structure

Please double check what version do you have locally.  
Also please check `yaml_cpp_vendor package if it is up to date and pulling correct version.

SimeonOA, 2022-07-29:
@MichaelOrlov 

I checked and the yaml_cpp_vendor_package being pulled is up to date. 

Moreover, this compile error with rosbag2 happens before yaml_cpp_vendor_package is built.

It is possible that another ros2 repo is using an older version of yaml_cpp but I have not seen any so far. 

The ros2 repos and dependencies are imported using ` vcs import src < ros2.repos` and seem to be imported and then built in alphabetical order. I tried changing the order in the ros2.repos file to import yaml_cpp_vendor_package before rosbag2,  thus ensuring that the yaml_cpp version would be up to date. This did not work as it seems vcstool imports in alphabetical order regardless.

MichaelOrlov, 2022-08-02:
@SimeonOA I would suggest to try to force building `yaml-cpp` from sources via following commands:
```bash
colcon build --packages-select yaml_cpp_vendor --cmake-args -DFORCE_BUILD_VENDOR_PKG=ON
```
Then try to build `rosbag2_storage` package
```bash
source ./install/setup.bash
colcon build --packages-select rosbag2_storage
```
It should pull out correct version via internet, build and install it in local `install` folder.
It looks like by default `yaml_cpp_vendor` package taking `yaml-cpp` version which he found installed in the system . And it's likely outdated on your system.
https://github.com/ros2/yaml_cpp_vendor/blob/16f9d254a3107b9b03a6805e6e5e51f768effb44/CMakeLists.txt#L69-L88

MichaelOrlov, 2023-04-13:
- Closing as stale issue with no response from author.

closed 2023-04-13 by MichaelOrlov

acochrane, 2023-12-21:
I had this error with the YAML library on jetson tx2, but the procedure suggested

> @SimeonOA I would suggest to try to force building `yaml-cpp` from sources via following commands:

was not a solution.

I got successful colcon build after removing the apt package libyaml-cpp-dev, suggested by @manato [here](https://github.com/dusty-nv/jetson-containers/issues/160).
Maybe it was first in the cmake search list?