ros2/rosbag2 #1048: Weird filepath behavior with `rosbag2_cpp/writer`

AndyZe, 2022-07-22:
I'm trying to write a bag like this. It does work :+1: but the directory/file naming behavior is strange. I cribbed heavily from the [unit tests](https://github.com/ros2/rosbag2/blob/master/rosbag2_tests/test/rosbag2_tests/test_rosbag2_cpp_api.cpp#L49-L87).

```
  std::string filepath = "/home/andy/Documents/test";
  rosbag2_cpp::Writer writer;
  try
  {
    writer.open(filepath);
  }
  ...
```

This creates the folder and saves the file in `/home/andyz/Documents/test/test_0.db3`

The weird behavior is:

1) If the `/test` folder already exists, writing fails. The error is:

```
[task_planning_action_server-1] terminate called after throwing an instance of 'std::runtime_error'
[task_planning_action_server-1]   what():  Database directory already exists (/home/andy/Documents/test), can't overwrite existing database
```

2) The prefix of the filename takes the folder name. `test_0.db3`. I don't see why the file name would have anything to do with the folder name. Seems like it should be automatically determined by timestamp or a separate argument should specify the filename.

IMO the behavior should be:

1) If the `/test` folder already exists, that's fine
2) If there is another file with the same name already, the new bag file name gets incremented. `test_0`, `test_1`, `test_2`, etc

EsipovPA, 2022-07-22:
Sorry for intrusion, by why not use temporary files, or folders for testing? If the use case involves recording consecutive bags, maybe it will be easier to include recording start time to the path, to avoid path intersections?

Something like this, for example:
/path/to/bags/{start_timestamp}/test

AndyZe, 2022-07-23:
Well, that is what we're doing for now.  :+1:

MichaelOrlov, 2022-07-26:
@AndyZe This is not a bug, this is a feature :wink: 

In rosbag2 by design each recording supposed to be in separate folder.  
Writer accept storage options as input parameter for open method.
StorageOptions->uri is a path to the base folder which rosbag2 writer will create by its own and manage recording files inside of it.

We are generating folder name with current timestamp in python layer which is parsing command line arguments
https://github.com/ros2/rosbag2/blob/5dc9e7f9a7a940815259f1d11785d10a13e14ca7/ros2bag/ros2bag/verb/record.py#L194-L197

This design decision with own folder was made since additionally we are writing `metadata.yaml` file after each recording. Also we have feature for splitting bags by maximum size. 
i.e. to avoid some mess we decided to have one folder per one recording with our own file structures insight. User shall not try to manage files inside this folder manually.
For playback we usually specify path to the folder where recorded files are. Not to the dedicated `db` file. And player managing by itself how to play multiple files from those folder.

closed 2022-07-26 by MichaelOrlov