ros2/rosbag2 #1049: tf_static doesnt work with rosbag2 but tf does

saippua, 2022-07-27:
## Description
I'm saving a rosbag2 with `ros2 bag record -a`, playing it back and viewing it in rviz2.
I'm publishing a pose and a transformation. the transformation is static (it doesnt change)

### Expected Behavior
When I playback the rosbag2, I should be able to see the transformation in rviz2.

### Actual Behavior
* if I publish the transformation using `StaticTransformBroadcaster` (once at the beginning of the recording), the transformation does not appear in rviz2 when I play the bag. I've checked that the bag file has the `/tf_static` topic and there is a message there.
* If I publish the transformation using `TransformBroadcaster`, the transformation *does* appear in rviz2, and I see the tf tree correctly.
* If I run rviz2 while publishing to visualize the data feed directly without using rosbag2 in the middle, both `StaticTransformBroadcaster` and `TransformBroadcaster` work.

## To Reproduce
1. run a publisher that publishes a pose and transformation
2. record it using `ros2 bag record -a`
3. play back the recording `ros2 bag play -l <bag>`
4. view result in rviz2. If using `StaticTransformBroadcaster`, the tf wont be in rviz.

## System (please complete the following information)
 - OS: Ubuntu 20.04
 - ROS 2 Distro: Galactic
 - Version: rosbag2 0.9.1

MichaelOrlov, 2022-07-27:
@saippua Can you please provide example of the rosbag2 with messages from `StaticTransformBroadcaster` which are not displaying in rviz?

bjajoh, 2022-10-03:
I have exactly the same issue, static tf transforms are not being able to be visualized with rviz.

MichaelOrlov, 2022-10-03:
@saippua @bjajoh We need reliable  reproducer for this issue. 
Could you please provide recorded bags with `StaticTransformBroadcaster` and with `TransformBroadcaster` to be able to reproduce and debug issue or please provide minimum valuable reproducer for `item 1`.
> 1. run a publisher that publishes a pose and transformation

FlorianPix, 2023-02-15:
I have the same issue. 
[Heres how the tf tree should look like.](https://github.com/ros2/rosbag2/files/10741533/normal.pdf)
[Here's how it looks when replaying the ros2 bag.](https://github.com/ros2/rosbag2/files/10741535/ros2_bag.pdf)

Alex-Beh, 2023-03-02:
Hello, I am facing the same issue on ROS2 humble & foxy. Will try to debug and report back if there is any findings.

MichaelOrlov, 2024-05-05:
- Closing as a stale issue.

closed 2024-05-05 by MichaelOrlov