ros2/rosbag2 #1055: ros2 bag play under docker container with large rosbags

ghanim-mukhtar, 2022-08-01:
## Description
So I am running ros2 bag play for rosbags that have been recorded with ros2 bag record and all this under a docker container that run ROS 2 Foxy focal with Ubuntu 20.04 focal.

### Expected Behavior
To load and play the rosbags till the end. 

### Actual Behavior
If we are playing rosbags that are small enough (this is relative but like 5 GB) it works fine till the end but for large files it starts and then get killed and I look at `htop` in the native machine and it seems it is an issue of memory (swap). An example of the output is:
```
[INFO] [1659367143.338281609] [rosbag2_storage]: Opened database 'recording/rosbag_azure_test_externaldisk//rosbag_azure_test_externaldisk_0.db3' for READ_ONLY.
Killed
```

## To Reproduce
1. Create a Docker container based on `ros:foxy-ros-base-focal` then build and run it.
2. Record huge rosbag file (I used kinect azure camera with `ros2 bag record -a`)
3. Replay it.

## System (please complete the following information)
 - OS: Ubuntu 20.04 focal (native and in the docker container)
 - ROS 2 Distro: foxy focal

## Additional context
The size of the swap in my machine (DELL Latitude) is 2 GB

MichaelOrlov, 2022-08-02:
@ghanim-mukhtar Can you please try to play with command line option `--read-ahead-queue-size 100` or `--read-ahead-queue-size 10` ?

MichaelOrlov, 2022-08-02:
@ghanim-mukhtar You also can provide detailed `SQLite` configuration via pragmas https://sqlite.org/pragma.html.
To specify pragmas will need to create storage config file. Please refer to the https://github.com/ros2/rosbag2/blob/a3b461c058a820ef4b47db480d1c84bf8921214b/rosbag2_performance/rosbag2_performance_benchmarking/config/storage/storage_resilient.yaml as example.

To be able use custom storage config file need specify path to the storage config file via `--storage-config-file` command line option for `ros2 bag play` command.

ghanim-mukhtar, 2022-08-02:
@MichaelOrlov yes with the parameters `--read-ahead-queue-size` set to 100 it worked, thank you for the tip. FYI I had a warning about rate starving and ask me to increase the rate but anyway the play was done till the end. 

Also I want to mention that there is no param `--storage-config-file` here is the output of `--help` with `play` command:
```
ros@delta-vision-ros2:~/delta_vision_ws$ ros2 bag play --help
usage: ros2 bag play [-h] [-s STORAGE] [--read-ahead-queue-size READ_AHEAD_QUEUE_SIZE] [-r RATE] [--topics TOPICS [TOPICS ...]] [--qos-profile-overrides-path QOS_PROFILE_OVERRIDES_PATH] [-l]
                     [--remap REMAP [REMAP ...]]
                     bag_file

Play back ROS data from a bag

positional arguments:
  bag_file              bag file to replay

optional arguments:
  -h, --help            show this help message and exit
  -s STORAGE, --storage STORAGE
                        storage identifier to be used, defaults to 'sqlite3'
  --read-ahead-queue-size READ_AHEAD_QUEUE_SIZE
                        size of message queue rosbag tries to hold in memory to help deterministic playback. Larger size will result in larger memory needs but might prevent delay of message playback.
  -r RATE, --rate RATE  rate at which to play back messages. Valid range > 0.0.
  --topics TOPICS [TOPICS ...]
                        topics to replay, separated by space. If none specified, all topics will be replayed.
  --qos-profile-overrides-path QOS_PROFILE_OVERRIDES_PATH
                        Path to a yaml file defining overrides of the QoS profile for specific topics.
  -l, --loop            enables loop playback when playing a bagfile: it starts back at the beginning on reaching the end and plays indefinitely.
  --remap REMAP [REMAP ...], -m REMAP [REMAP ...]
                        list of topics to be remapped: in the form "old_topic1:=new_topic1 old_topic2:=new_topic2 etc."
```
And here is the output of `sudo apt-cache policy ros-foxy-rosbag2`:
```
ros@delta-vision-ros2:~/delta_vision_ws$ sudo apt-cache policy ros-foxy-rosbag2
ros-foxy-rosbag2:
  Installed: 0.3.9-1focal.20220209.163828
  Candidate: 0.3.9-1focal.20220209.163828
  Version table:
 *** 0.3.9-1focal.20220209.163828 500
        500 http://packages.ros.org/ros2/ubuntu focal/main amd64 Packages
        100 /var/lib/dpkg/status
```

MichaelOrlov, 2022-08-02:
@ghanim-mukhtar `--storage-config-file` command line option available on `foxy-future` branch
https://github.com/ros2/rosbag2/blob/foxy-future/ros2bag/ros2bag/verb/play.py#L63-70

Please next time ask your questions on https://answers.ros.org/questions/ first

I am closing this issue as question is answered and no real issues in rosbag2.

closed 2022-08-02 by MichaelOrlov

fnobis, 2022-12-14:
@MichaelOrlov coming back to this: I replay a large rosbag with `ros2 bag play BAG_NAME -s map --read-ahead-queue-size 100`. Over time the used memory size increases until it is almost full (28/31GB). Then the computer hangs for a bit and the replay is terminated with the output "Killed". To me, it seems there is indeed an issue with the playback currently?

MichaelOrlov, 2022-12-16:
@fnobis Thanks for letting us know about this problem. 
To be able to move forward with issue analysis we need to be able to reproduce this issue.
Can you please provide minimum valuable reproducer or exact steps how to reproduce this problem?

fnobis, 2022-12-23:
@MichaelOrlov just a quick status: I tried to re-create a similar a bag file with a script. So far, I haven't been able to get the same effect with this artificial data. I hope I'll have some time the coming days to replicate the original bag in more detail, so that I can provide a script for this. It does not seem to happen for every bag, just for the recording I mentioned above.

karanchahal, 2024-05-24:
Hi I experience this prpblem with very large rosbags, around 30 GB size, the serialization holds onto memory and just doesnt clear it even if the objects I am using are not longer used. I am using the python API btw, I dont use the RMW at all here,