ros2/rosbag2 #1056: ros2 bag play uses wall time?

poornimajd, 2022-08-02:
Hello all,
I have recorded the bag file in foxy distribution and while playing the bag file in galactic I use the following command-
`ros2 bag play name_of_the_bag_file`.
I then try to run the nav2 stack and it runs without any errors.
But does this mean that the above command is using the Wall time of the system and hence the topics published from the bag file have the current wall time of the system?
I also tried with the option 
`ros2 bag play name_of_the_bag_file --clock`.
But the nav2 stack again runs smoothly, but when we specify the clock option, the bag file is supposed to publish the topics with the time-stamped when it was recorded right, and since the nav2 stack is running on the wall time, there should be a TF error.
Basically in both of the above cases, there was no difference in the operation of the nav2 stack.
So what is the option to run the bag file so that it takes the wall time of the system?
Could anyone please suggest or clarify the above question?
Any suggestion is greatly appreciated!

vinnnyr, 2022-08-03:
how are you running the nav2 stack? depending on how you are running it, `use_sim_time` might be set to `true` for all the launch, hence you would be getting expected behavior.

poornimajd, 2022-08-04:
Hello, Thank you for the reply.
I have set use_sim_time=False. 
Actually, when I play the bag file, the time stamps of the topics are the same as the day when it was recorded.
The nav2 stack takes these topics as input and the timestamp of the published pose (amcl_pose) is also the same as the timestamp of the bag file, Hence both the nav2stack and the bag file are using the same timestamp, but it is not the current wall clock of the system, it is the time when the bag file was recorded.
So is it possible to play the bag file to take the current wall clock of the system for the timestamp of the topics? It did not make any difference when I also used the --clock option.
I am not using any simulation, these are all physical experiments.

MichaelOrlov, 2022-09-10:
@poornimajd Short answer is NO. 
Rosbag2 player uses timestamp from recorded messages.
Currently we don't have option to use wall time for playing messages.

The reason for that is that message itself coming to the rosbag2 recorder in serialized form and when we publish it in player we publishing it as is directly in raw serialized form without transforming it to the ROS message. 
Therein no place where we can change timestamp for the publishing message.

closed 2022-09-10 by MichaelOrlov