ros2/rosbag2 #1061: Make binary release for rosbag2 from rolling 0.17.0 tag

MichaelOrlov, 2022-08-08:
I've tried to make release with bloom, however got error message about insufficient rights

```
bloom-release --rosdistro rolling rosbag2
ROS Distro index file associate with commit '0fa1904294ab891b243e231f14de8a995f3de3f9'
New ROS Distro index url: 'https://raw.githubusercontent.com/ros/rosdistro/0fa1904294ab891b243e231f14de8a995f3de3f9/index-v4.yaml'
==> Fetching 'rosbag2' repository from 'https://github.com/ros2-gbp/rosbag2-release.git'
Cloning into '/tmp/tmpwmey0xif'...
remote: Enumerating objects: 55104, done.
remote: Counting objects: 100% (4032/4032), done.
remote: Compressing objects: 100% (2947/2947), done.
remote: Total 55104 (delta 1181), reused 3676 (delta 944), pack-reused 51072
Receiving objects: 100% (55104/55104), 8.50 MiB | 6.60 MiB/s, done.
Resolving deltas: 100% (18329/18329), done.
==> Testing for push permission on release repository
==> git remote -v
origin	git@github.com:ros2-gbp/rosbag2-release.git (fetch)
origin	git@github.com:ros2-gbp/rosbag2-release.git (push)
==> git push --dry-run
ERROR: Permission to ros2-gbp/rosbag2-release.git denied to MichaelOrlov.
fatal: Could not read from remote repository.

Please make sure you have the correct access rights
and the repository exists.
```
It seems that I need to be in [ros2-gbp/rosbag2](https://github.com/ros2-gbp/ros2-gbp-github-org/blob/latest/rosbag2.tf) team to have permissions to push in those git repo.

I have created https://github.com/ros2-gbp/ros2-gbp-github-org/issues/90 with request to join `rosbag2` release team.

MichaelOrlov, 2022-08-08:
It's uncertain how long it will take to get a right permissions for me.
@emersonknapp Could you please make bloom release for rosbag2 rolling from tag 0.17.0 meanwhile?
It seems that you have necessary permissions for [ros2-gbp](https://github.com/ros2-gbp/ros2-gbp-github-org/blob/latest/rosbag2.tf) project.

emersonknapp, 2022-08-10:
Yes, I can do this today

MichaelOrlov, 2022-08-11:
Thanks @emersonknapp, I appreciate your help.
I am currently on my vacation and can’t do bloom release by myself.

emersonknapp, 2022-08-11:
https://github.com/ros/rosdistro/pull/34115

MichaelOrlov, 2022-08-16:
@emersonknapp I see that you have increased version of packages in [rolling/distribution.yaml](https://github.com/ros/rosdistro/pull/34115/files#diff-a79d50d14101a47dfc4ea7b0dd7b9446f34b544807d8f0c48cc6a8f07fc3b36aR4189) file up to the `0.17.0-2`.

I am a bit confused what is the meaning for `-2` at the end of release digits `0.17.0-2` ? 
I would expect to see the same digits as in tag `0.17.0`.

emersonknapp, 2022-08-19:
You will notice that all bloom releases have a "release inc" suffix on them, it is always `0.17.0-1` and so on. I don't know the details of how the release increment works, but it's an internal detail, and the released version is 0.17.0. The extra increment might have come from a failed bloom command on the first try when i ran it, then the second time increased that number. It doesn't affect the release details though.

emersonknapp, 2022-08-19:
You can see the history of commits auto-created by bloom at https://github.com/ros2-gbp/rosbag2-release/commits/master

clalancette, 2022-08-19:
> You will notice that all bloom releases have a "release inc" suffix on them, it is always `0.17.0-1` and so on. I don't know the details of how the release increment works, but it's an internal [...]

Right, the version number `0.17.0` always corresponds to the source release version (basically the tag in this git repository).

The extra `-1` is called the "debinc" number, and is essentially a count of how many times the package metadata was updated during the release (the original source remains the same, though the metadata can patch the source).

In our particular case, bloom is in charge of incrementing that number, and does so anytime you do a release where the source version is the same as in a previous release.  This can happen for a number of reasons, including a failed bloom-release, changes to the tracks/package metadata, etc.