ros2/rosbag2 #1062: No recording happening with a lot of topics

dbruggner, 2022-08-09:
## Description
When recording a lot of topics rosbag2 doesn't subscribe to all topics and data is not recorded.

### Expected Behavior
It is expected that rosbag2 will subscribe to all topics (even if these are a lot) and record all the data.

### Actual Behavior
This behavior can be reproduced with the following launch file. For example rosbag2 records fine with 10 `image_publishers` but not with 100 `image_publishers`. The log shows the output of `ros2 bag info` for a bag file recorded for 100 `image_publishers`
Launch file:
`import launch
from launch import LaunchDescription
from launch_ros.actions import Node
from launch.actions import ExecuteProcess
from launch.substitutions import LaunchConfiguration

from typing import Final, List
from datetime import datetime

NAMESPACE: Final[str] = "/my_space"

def generate_launch_description():

    nodes: Final[List[Node]] = []

    for i in range(100):
        nodes.append(
            Node(
                name="image_pub_" + str(i),
                executable="image_publisher_node",
                package="image_publisher",
                namespace=NAMESPACE + "/_" + str(i),
                parameters=[
                    {"publish_rate" : 20.0},
                    {"filename" : "/home/user/Pictures/Test.jpg"},
                    ],
                ))
    
    bag_record: Final[List[ExecuteProcess]] = [
        ExecuteProcess(
            cmd=["ros2", "bag", "record",  "-a", "--include-hidden-topics", "-o","/tmp/"+datetime.now().strftime("%Y%m%d_%H%M%S") ],
            output="screen",
            log_cmd=True,
        ),
    ]

    ld = LaunchDescription()

    ld.add_action(LaunchDescription(nodes))
    ld.add_action(LaunchDescription(bag_record))

    return ld`

Log for 100 `image_publishers`:
`Files:             20220809_093334_0.db3
Bag size:          889.9 KiB
Storage id:        sqlite3
Duration:          34.364s
Start:             Aug  9 2022 09:33:51.367 (1660037631.367)
End:               Aug  9 2022 09:34:25.732 (1660037665.732)
Messages:          1785
Topic information: Topic: /nps/_59/camera_info | Type: sensor_msgs/msg/CameraInfo | Count: 18 | Serialization Format: cdr
                   Topic: /nps/_61/camera_info | Type: sensor_msgs/msg/CameraInfo | Count: 11 | Serialization Format: cdr
                   Topic: /nps/_28/image_raw | Type: sensor_msgs/msg/Image | Count: 0 | Serialization Format: cdr
                   Topic: /nps/_13/camera_info | Type: sensor_msgs/msg/CameraInfo | Count: 5 | Serialization Format: cdr
                   Topic: /nps/_70/image_raw | Type: sensor_msgs/msg/Image | Count: 0 | Serialization Format: cdr
                   Topic: /nps/_32/image_raw | Type: sensor_msgs/msg/Image | Count: 0 | Serialization Format: cdr
                   Topic: /nps/_56/image_raw | Type: sensor_msgs/msg/Image | Count: 0 | Serialization Format: cdr
                   Topic: /nps/_38/camera_info | Type: sensor_msgs/msg/CameraInfo | Count: 1 | Serialization Format: cdr
                   Topic: /nps/_87/camera_info | Type: sensor_msgs/msg/CameraInfo | Count: 5 | Serialization Format: cdr
                   Topic: /nps/_49/image_raw | Type: sensor_msgs/msg/Image | Count: 0 | Serialization Format: cdr
[... 208 lines cut ...]
                   Topic: /nps/_3/camera_info | Type: sensor_msgs/msg/CameraInfo | Count: 237 | Serialization Format: cdr
                   Topic: /nps/_1/image_raw | Type: sensor_msgs/msg/Image | Count: 46 | Serialization Format: cdr
                   Topic: /nps/_7/image_raw | Type: sensor_msgs/msg/Image | Count: 94 | Serialization Format: cdr
                   Topic: /nps/_9/image_raw | Type: sensor_msgs/msg/Image | Count: 144 | Serialization Format: cdr
                   Topic: /nps/_0/image_raw | Type: sensor_msgs/msg/Image | Count: 185 | Serialization Format: cdr
                   Topic: /nps/_1/camera_info | Type: sensor_msgs/msg/CameraInfo | Count: 242 | Serialization Format: cdr
                   Topic: /nps/_6/camera_info | Type: sensor_msgs/msg/CameraInfo | Count: 233 | Serialization Format: cdr
                   Topic: /nps/_5/image_raw | Type: sensor_msgs/msg/Image | Count: 96 | Serialization Format: cdr
                   Topic: /nps/_3/image_raw | Type: sensor_msgs/msg/Image | Count: 87 | Serialization Format: cdr
                   Topic: /nps/_8/image_raw | Type: sensor_msgs/msg/Image | Count: 120 | Serialization Format: cdr
                   Topic: /nps/_8/camera_info | Type: sensor_msgs/msg/CameraInfo | Count: 241 | Serialization Format: cdr
                   Topic: /nps/_7/camera_info | Type: sensor_msgs/msg/CameraInfo | Count: 235 | Serialization Format: cdr
`

## To Reproduce
1. Install `ros-galactic-image-publisher`
2. Create a ROS package with the given launch file
3. `source your_workspace/setup.bash`
4. Run the launch file with `ros2 launch your_package the_launch_file.py
5. `ros2 bag info /tmp/2022*`

## System (please complete the following information)
 - OS: Ubuntu Focal
 - ROS 2 Distro: Galactic
 - Version: galactic

MichaelOrlov, 2022-08-16:
@dbruggner What is the size of the publishing image? 
And what DDS vendor (`FastRTPS`, `CycloneDDS` or `RTI Connext DDS`) did you use?

dbruggner, 2022-08-17:
The size of the publishing image can be anything. I tested also with 1 pixel sized images.
The used DDS is Cyclone.

dbruggner, 2022-08-17:
It seems that `get_topic_names_and_types` https://github.com/ros2/rosbag2/blob/rolling/rosbag2_transport/src/rosbag2_transport/recorder.cpp#L262 returns `0` most of the times when a lot of topics are available. It happens also when I want to record only one topic.

MichaelOrlov, 2022-08-17:
@dbruggner We have came across with the same issue some time ago in Apex.OS.  
After comprehensive analysis we found out that root cause of the problem not in the `rosbag2` itself rather in CycloneDDS.

`get_topic_names_and_types()` will only return max 100 topics, because we are using graph info inside where we read the samples (and not take), the CycloneDDS read API sets the max samples to 100 internally [here](https://github.com/eclipse-cyclonedds/cyclonedds/blob/c1f7ed4c2bd14542413b24406072c2a8b903f2b0/src/core/ddsc/src/dds_read.c#L222) which causes this issue.

MichaelOrlov, 2022-08-17:
Closing this issue since this is not a bug in rosbag2 but rather "feature"/bug on CycloneDDS side.

closed 2022-08-17 by MichaelOrlov