ros2/rosbag2 #1075: Message listener for recording

emersonknapp, 2022-08-20:
## Description

Allow user to register a callback with the Recorder that will be called on every new recorded message. This need comes out of `rqt_bag` workflows. See #558 for full discussion.

## Related Issues

Split from #558

## Completion Criteria

* Add a user-provided callback to Recorder that can be called on every new message recorded
* Expose this API to Python client as well

## Implementation Notes / Suggestions

** If you have ideas about how this feature might be accomplished, put them here. Note that this is just a suggestion to the implementer, so feel free to speculate. **

## Testing Notes / Suggestions

** All features in this project need tests. Please give some input on cases that will need to be tested - and how the testing might be implemented. **

MichaelOrlov, 2022-09-01:
@emersonknapp I am afraid that callback from each new message to the python API and then to the rqt_bag will hit performance of the entire recording process.  
I think it would be better to gather needed statistics inside recorder and provide getters API  to it. Perhaps we can use atomic variables without mutexes. 
The idea is that `rqt_bag` will call statistic's getter API periodically instead of subscribing to the callbacks for each message write.

emersonknapp, 2022-09-02:
Well, historically, `rqt_bag` has used a much more inefficient method - when a message comes in, it triggers a separate thread to ask the actual written bagfile for messages in a timestamp range, and uses those results to populate the timeline. I'm not saying this is good, but a callback would be a strict improvement on running a read query on the bag that is being written, for every new message.

emersonknapp, 2025-02-21:
I'm going to close this as not planned, since rqt_bag maintenance seems relatively stalled.

closed 2025-02-21 by emersonknapp