ros2/rosbag2 #1079: ros2 bag record does not properly close recoring

FranzAlbers, 2022-09-05:
## Description
When closing a recording, occasionally, the ros2 bag is apparently not closed correctly, which makes a reindex of the bag necessary. 
This happens for both the sqlite3 and the mcap plugins. Usually, this is no big issue and takes only a few seconds, but reindexing these mcap files seems to be not possible for now (which is another issue).
For our setup (recording point clouds, images, and GNSS data), this issue occurs in most cases.

### Expected Behavior
The recording should stop and write the remaining messages from the cache to the bag:
`...
[INFO] [1662381739.854492546] [rosbag2_recorder]: Subscribed to topic '/sensing/gnss/fixed'
[INFO] [1662381739.856257118] [rosbag2_recorder]: Subscribed to topic '/sensing/gnss/pose'
[INFO] [1662381746.062154306] [rclcpp]: signal_handler(signum=2)
[INFO] [1662381746.159286989] [rosbag2_cpp]: Writing remaining messages from cache to the bag. It may take a while
`

### Actual Behavior
ros2 bag record is stuck at `signal_handler(signum=2)`. The remaining messages are not written to the bag and killing the recording process or closing the terminal is required. 
`...
[INFO] [1662381748.073734704] [rosbag2_recorder]: Subscribed to topic '/sensing/gnss/fixed'
[INFO] [1662381748.076060419] [rosbag2_recorder]: Subscribed to topic '/sensing/gnss/pose'
[INFO] [1662381754.862466740] [rclcpp]: signal_handler(signum=2)
`

## To Reproduce
** Steps to reproduce the behavior, e.g.
1. Start a recording with `ros2 bag record -a`. Actually, this issue also occurs when no other node is running and only the recording is started.
2. Stop recording with Ctrl+C.
3. It might need a few tries, but for me, in roughly half of the cases the recording is not stopped correctly, and killing the process is required.

## System (please complete the following information)
 - OS: Ubuntu Jammy
 - ROS 2 Distro: Humble
 - Version: 0.15.2-1jammy.20220729 (apt release)

## Additional context
** Add any other context about the problem here **

MichaelOrlov, 2022-09-07:
@FranzAlbers It seems that I know what is causing this issue and I have a fix for it.  
To double check can you please try to press `Enter` key when you see that rosbag2 reorder hangout after `Ctrl+C`?

Sorry for inconvenience. I am expecting that pressing `Enter`  should work as workaround until we will fix this issue with a proper way.

FranzAlbers, 2022-09-07:
@MichaelOrlov, thanks for your help and for the hint! 

Pressing enter indeed works as a workaround.

PR #1081 (2022-09-10): Fix hangout in rosbag2 player and recorder when pressing `CTRL+C`
- Closes #1079 
We don't need to explicitly handle `SIGINT` in `KeyboardHandler` instantiated from rosbag2 executables since we have already overriding `SIGINT` on upper python level or inside `rclcpp::init()` and correctly finishing application with proper calling of all destructors upon receiving `SIGINT` via checking `rclcpp::ok()` method in infinite loop.

Please note that it's almost impossible to cover this fix in unit tests since `gtest` using redirection of the standard input to the file and `KeyboardHandler` has detecti [... cut]
closed 2022-09-10 by MichaelOrlov