ros2/rosbag2 #1080: MultiLibraryClassLoader failed

lhsl2014, 2022-09-06:
## Description
Using the Galactic and rosbag2 to parse deb3 bag.  The rosbag2_cpp::readers::SequentialReader works well with one deb3 bag, but sometimes failed when I trying to open two or more deb3 bags at the same time. The rosbag2_storage shows message "Unable to load instance of read write interface: MultiLibararyClassLoader: Could ont create class of type rosbag2_storage_plugins::SqliteStorage".

This situation appear not every time but sometimes. This really confuse me.

lhsl2014, 2022-09-28:
Can someone help with this?

MichaelOrlov, 2022-09-28:
@lhsl2014 Could you please  provide minimum valuable reproducer or steps how to reproduce this issue?

lhsl2014, 2022-09-29:
Thank you for your reply. In my case, I use the std::thread to deserialize the Camera and Lidar deb3 bag at the same time. The code is listed as follow.
##code
void camera_thread()
{
rosbag2_cpp::readers::SequentialReader reader_cam;

rosbag2_storage::StorageOptions storage_options{};
storage_options.uri  = "./";
storage_options.storage_id="sqlite3";

rosbag2_cpp::ConverterOptions converter_options{};
converter_options.input_serialization_format = "cdr";
converter_options.output_serialization_format = "cdr";

reader_cam.open(storage_options,converter_options);

while(reader_cam.has_next())
{...}
}
void lidar_thread()
{
ros2bag2_cpp::readers::SequentialReader reader_lidar;

rosbag2_storage::StorageOptions storage_options{};
storage_options.uri  = "./";
storage_options.storage_id="sqlite3";

rosbag2_cpp::ConverterOptions converter_options{};
converter_options.input_serialization_format = "cdr";
converter_options.output_serialization_format = "cdr";

reader_lidar.open(storage_options,converter_options);

while(reader_lidar.has_next())
{...}
}

void main()
{
...
std::thread camera_th(camera_thread);
std::thread lidar_th(lidar_thread);

camera_th.join();
lidar_th.join();

...
}
##

When running the code to read two deb3 bag, the rosbag2_storage shows message "Unable to load instance of read write interface: MultiLibararyClassLoader: Could ont create class of type rosbag2_storage_plugins::SqliteStorage" and quit abnormal.

lhsl2014, 2022-09-29:
@MichaelOrlov Thank you for your reply. In my case, I use the std::thread to deserialize the Camera and Lidar deb3 bag at the same time. The code is listed as follow.
## code
void camera_thread()
{
rosbag2_cpp::readers::SequentialReader reader_cam;

rosbag2_storage::StorageOptions storage_options{};
storage_options.uri = "./";
storage_options.storage_id="sqlite3";

rosbag2_cpp::ConverterOptions converter_options{};
converter_options.input_serialization_format = "cdr";
converter_options.output_serialization_format = "cdr";

reader_cam.open(storage_options,converter_options);

while(reader_cam.has_next())
{...}
}
void lidar_thread()
{
ros2bag2_cpp::readers::SequentialReader reader_lidar;

rosbag2_storage::StorageOptions storage_options{};
storage_options.uri = "./";
storage_options.storage_id="sqlite3";

rosbag2_cpp::ConverterOptions converter_options{};
converter_options.input_serialization_format = "cdr";
converter_options.output_serialization_format = "cdr";

reader_lidar.open(storage_options,converter_options);

while(reader_lidar.has_next())
{...}
}

void main()
{
...
std::thread camera_th(camera_thread);
std::thread lidar_th(lidar_thread);

camera_th.join();
lidar_th.join();

...
}

When running the code to read two deb3 bag, the rosbag2_storage shows message "Unable to load instance of read write interface: MultiLibararyClassLoader: Could ont create class of type rosbag2_storage_plugins::SqliteStorage" and quit abnormal.

lhsl2014, 2022-09-29:
@MichaelOrlov Thank you for your reply. In my case, I use the std::thread to deserialize the Camera and Lidar deb3 bag at the same time. The code is listed as follow.

## code
void camera_thread()
{
rosbag2_cpp::readers::SequentialReader reader_cam;

rosbag2_storage::StorageOptions storage_options{};
storage_options.uri = "./";
storage_options.storage_id="sqlite3";

rosbag2_cpp::ConverterOptions converter_options{};
converter_options.input_serialization_format = "cdr";
converter_options.output_serialization_format = "cdr";

reader_cam.open(storage_options,converter_options);

while(reader_cam.has_next())
{...}
}
void lidar_thread()
{
ros2bag2_cpp::readers::SequentialReader reader_lidar;

rosbag2_storage::StorageOptions storage_options{};
storage_options.uri = "./";
storage_options.storage_id="sqlite3";

rosbag2_cpp::ConverterOptions converter_options{};
converter_options.input_serialization_format = "cdr";
converter_options.output_serialization_format = "cdr";

reader_lidar.open(storage_options,converter_options);

while(reader_lidar.has_next())
{...}
}

void main()
{
...
std::thread camera_th(camera_thread);
std::thread lidar_th(lidar_thread);

camera_th.join();
lidar_th.join();

...
}

## Report
When running the code to read two deb3 bag, the rosbag2_storage shows message "Unable to load instance of read write interface: MultiLibararyClassLoader: Could ont create class of type rosbag2_storage_plugins::SqliteStorage" and quit abnormal.

MichaelOrlov, 2022-09-29:
@lhsl2014  If I understand correctly you trying to create two instances of the `ros2bag2_cpp::readers::SequentialReader` in two separate thread which are opening different bags ans at some time you are getting exception with message `"Unable to load instance of read write interface: MultiLibararyClassLoader: Could ont create class of type rosbag2_storage_plugins::SqliteStorage"`.

From the first glance it looks like some concurrency problem in `MultiLibararyClassLoader` or `SequentialReader` constructor.

I can suggest a quick workaround in creating readers in advance before starting threads and pass them to target threads via parameter of private member variables.

lhsl2014, 2022-10-10:
> 

Thank you for your reply. We solved the problem according to your advice.

closed 2022-10-10 by lhsl2014