ros2/rosbag2 #1087: Ability to split logs based on signal during record.

rshanor, 2022-09-12:
## Description

I am setting up a robot that does pick and place tasks. I would like one rosbag associated with each pick and place action. I would like to be able to signal to the rosbag process to close the current bag and open the new one. I could probably work around this by killing the `ros2 bag record` process and starting a new one.

## Related Issues

None that I found.

## Implementation Notes / Suggestions

If this feature sounds useful, I can probably help implement. I imagine this takes the form of something like a `SplitLog.srv`. There are probably some nuances around how the logs are split to make sure data does not get lost on log boundaries.

vinnnyr, 2022-09-21:
Could we piggy back off of `SnapshotMode` and simply split off the bag every time the snapshot service is called in this special mode?

MichaelOrlov, 2022-09-23:
@rshanor @vinnnyr The proper design for this feature would be to create new service call handler `SplitBagfile.srv` in rosbag2 recorder class and promote `SequentialWriter::split_bagfile()` method to the `BaseWriterInterface` as virtual public method to be able to call it from recorder upon receiving service call for split.

rshanor, 2022-10-05:
Working to implement @MichaelOrlov suggestion.

closed 2022-10-12 by MichaelOrlov