ros2/rosbag2 #1095: Exclude/include topics by type

tonynajjar, 2022-09-20:
## Description

** Clear and concise description of the feature that you'd like to exist, from a user's perspective. Don't get into implementation details, describe what you want it to do. Put any consideration of alternate features that might solve your problem as well. **

We can already exclude topics by name, it would be nice to exclude topics by type. For my case, I'd like to avoid recording any Pointcloud2 to avoid an accidental size increase 

## Related Issues

[** Is this feature dependent on any other features? Is it part of a larger project? Note here. **
](https://github.com/ros2/rosbag2/issues/560)

## Completion Criteria

** What needs to be true before we can call this "Done"? Bullet lists are appropriate. **

As an example: `ros2 bag record -all --exclude-types sensor_msgs/msg/PointCloud2`

## Implementation Notes / Suggestions

** If you have ideas about how this feature might be accomplished, put them here. Note that this is just a suggestion to the implementer, so feel free to speculate. **

## Testing Notes / Suggestions

** All features in this project need tests. Please give some input on cases that will need to be tested - and how the testing might be implemented. **

tonynajjar, 2024-04-10:
This is solved with https://github.com/ros2/rosbag2/pull/1582?

ahcorde, 2024-04-10:
I think so. PRs are merged on rolling

MichaelOrlov, 2024-04-10:
- Closing as resolved in the #1582

closed 2024-04-10 by MichaelOrlov