ros2/rosbag2 #1116: Track "default storage implementation" in a single place, to be used by other code

emersonknapp, 2022-10-06:
## Description

Right now there are a variety of places in the rosbag2 core that encode the explicit string `"sqlite3"`. This is a tight coupling between the core and the default storage implementation.

We should define in a single place what the default storage is, and use that wherever a default is required. Right now, this means finding all places where sqlite3 is mentioned explicitly, outside the sqlite3 storage implementation, and changing them so that it is not needed.

## Related Issues

## Completion Criteria

* No explicit mention of "sqlite3" string anywhere in rosbag2 core
* Allow empty storage id for record/write - Writer uses default storage implementation when not specified

## Implementation Notes / Suggestions

Two options I can think of
* Somehow use the `ament_index` to register a file from `rosbag2_storage_default_plugins`, which contains the name of the default plugin. This can be loaded by upstream packages
* Make `rosbag2_storage_default_plugins` provide a C++ header with a const string of the default plugin ID, which downstream packages can read and use. Expose this via `rosbag2_py` as well.

## Testing Notes / Suggestions

* All tests should run against the default storage implementation

james-rms, 2022-10-21:
@emersonknapp I might be missing something important, but the only places I can see `sqlite3` being hardcoded in this repo are:

* at the top of the `record` verb, here: https://github.com/ros2/rosbag2/blob/19158f6debe25640507881f9a2c01cdb1fe4741b/ros2bag/ros2bag/verb/record.py#L38
* All over the help strings / description of `ros2bag` verbs
* at the top of `writer.cpp`, here: https://github.com/ros2/rosbag2/blob/rolling/rosbag2_cpp/src/rosbag2_cpp/writer.cpp#L41

The rest seem to be part of test fixtures or performance evaluation code. Is there somewhere else I should be looking?

james-rms, 2022-11-02:
Plan: 

1. Expose a `get_default_storage_id()` function from `rosbag2_storage`, bind with pybind11
2. Use this to set the default for `ros2 bag record -s`, and the default for `Writer::open(std::string uri)`
3. for the tests, do a pass and split them between "sqlite3 specific" and "storage plugin agnostic". Leave the sqlite3-specific tests alone,  and parametrize the storage plugin agnostic tests across all default-installed plugins (sqlite3, and soon MCAP).

PR #1724 (2024-06-22): [humble] rosbag2_storage: expose default storage ID as method (backport #1146)

Fixes #1116 <hr>This is an automatic backport of pull request #1146 done by [Mergify](https://mergify.com).
closed 2022-11-09 by emersonknapp