ros2/rosbag2 #1119: `ros2 bag convert` with compression can drop messages

james-rms, 2022-10-10:
## Description
`ros2 bag convert` with message-level compression will silently write only some messages to the output bag.

### Expected Behavior
`ros2 bag convert` will convert bags at slower-than-real-time if neccessary, blocking on a full queue (rather than dropping messages).

Even if this can't be done, I would expect to see a warning if i'm dropping messages while converting.

### Actual Behavior
ROS2 bag convert appears to drop messages.

## To Reproduce
Grab a sample MCAP bag here: https://github.com/foxglove/mcap/blob/main/testdata/mcap/demo.mcap (uses ROS1 messages, but `ros2 bag convert` does not care)
```bash
$ cat << EOF > convert.yaml
output_bags:
  - uri: message_compressed_sqlite3
    compression_mode: message
    compression_format: zstd
    all: true
EOF
$ ros2 bag convert -i demo.mcap -o convert.yaml
# observe difference in message count
$ ros2 bag info demo.mcap

closing.

Files:             demo.mcap
Bag size:          58.6 MiB
Storage id:        mcap
Duration:          7.780s
Start:             Mar 22 2017 13:26:20.103 (1490149580.103)
End:               Mar 22 2017 13:26:27.884 (1490149587.884)
Messages:          1606
Topic information: Topic: /velodyne_points | Type: sensor_msgs/PointCloud2 | Count: 78 | Serialization Format: ros1
                   Topic: /radar/tracks | Type: radar_driver/RadarTracks | Count: 156 | Serialization Format: ros1
                   Topic: /radar/range | Type: sensor_msgs/Range | Count: 156 | Serialization Format: ros1
                   Topic: /radar/points | Type: sensor_msgs/PointCloud2 | Count: 156 | Serialization Format: ros1
                   Topic: /tf | Type: tf2_msgs/TFMessage | Count: 774 | Serialization Format: ros1
                   Topic: /image_color/compressed | Type: sensor_msgs/CompressedImage | Count: 234 | Serialization Format: ros1
                   Topic: /diagnostics | Type: diagnostic_msgs/DiagnosticArray | Count: 52 | Serialization Format: ros1
# missing many messages
$ ros2 bag info message_compressed_sqlite3
Files:             message_compressed_sqlite3_0.db3
Bag size:          4.3 MiB
Storage id:        sqlite3
Duration:          7.780s
Start:             Mar 22 2017 13:26:20.103 (1490149580.103)
End:               Mar 22 2017 13:26:27.884 (1490149587.884)
Messages:          62
Topic information: Topic: /diagnostics | Type: diagnostic_msgs/DiagnosticArray | Count: 2 | Serialization Format: ros1
                   Topic: /image_color/compressed | Type: sensor_msgs/CompressedImage | Count: 8 | Serialization Format: ros1
                   Topic: /radar/points | Type: sensor_msgs/PointCloud2 | Count: 8 | Serialization Format: ros1
                   Topic: /radar/range | Type: sensor_msgs/Range | Count: 7 | Serialization Format: ros1
                   Topic: /tf | Type: tf2_msgs/TFMessage | Count: 29 | Serialization Format: ros1
                   Topic: /velodyne_points | Type: sensor_msgs/PointCloud2 | Count: 8 | Serialization Format: ros1
```

Using `compression_queue_size: 1000` alleviates this issue for this bag, but not clear if this is sufficient for all bags.
## System (please complete the following information)
 - OS: Ubuntu 22.04 ARM64 running in Parallels on an M1 Macbook
 - ROS 2 Distro: Rolling
 - Version: 626d69bafd9a8fb922dc2ec9cb59c80d00e6d56f

amacneil, 2022-10-10:
Possible dupe: https://github.com/ros2/rosbag2/issues/998

MichaelOrlov, 2022-10-10:
@james-rms @amacneil I would say this is duplicate for https://github.com/ros2/rosbag2/pull/1058 and we have already have workaround for this.

The workaround is to set `compression_queue_size = 0`  in writer settings in yaml config file.

amacneil, 2022-10-11:
It is good that there is a workaround, but I don't think there should be any scenario in which `ros2 bag convert` can drop messages, regardless of what options the user passes. If it does, that should be an error because the bag failed to convert.

PR #1121 (2022-10-12): rosbag2_py: set defaults for config when bag rewriting

Fixes #1119
closed 2022-10-12 by MichaelOrlov