ros2/rosbag2 #1120: use rosbag write topic info, frame rate not stable

ChunelFeng, 2022-10-10:
## Description
write topic (10Hz) info with rosbag, but the rate maybe not always be 100ms.
it seemed  an about 2~3 second timespan sometimes happened.

### Expected Behavior
no write timespan.

### Actual Behavior
it seemed to happened, occasionally

## Additional context
can you give me some help about how to locate and fix this question.

MichaelOrlov, 2022-10-10:
@ChunelFeng Need more info about setup and reproduction steps.
Do you use bag split during recording?

ChunelFeng, 2022-10-19:
No，we no use bag spilt.

Just one bag, sometimes it happens

MichaelOrlov, 2022-10-19:
@ChunelFeng 2-3 seconds time difference for messages  with 10 Hz is to much.
No one yet observed such issue with rosbag2 recording in regular scenario.

All similar cases which was reported was caused ether by problem on publisher side or QoS settings as best effort for topic and heavy load on the system.

MichaelOrlov, 2022-10-29:
@ChunelFeng Please provide minimum valuable reproducer or exact steps how to reproduce this issue.
Otherwise we will have to close this issue as not reproducible.

closed 2022-10-30 by ChunelFeng