ros2/rosbag2 #1125: Using the default QoS yaml throws an error when playing the bag

geoeo, 2022-10-14:
## Description
Using the default QoS yaml on this site throws an error

### Expected Behavior
Using the default QoS Yaml given in the Readme.md as a offered_qos_profiles argument does not result in a crash when playing the bag

### Actual Behavior
Currently when using the default QoS Yaml I get a crash when playing my recorded bag `yaml-cpp: error at line 1, column 1: bad conversion`

## To Reproduce
1.Copy the QoS yaml and use it as `offered_qos_profiles' 
2. Record your bag
3. replay the bag via `ros2 bag play ..`
4. An error is given in the terminal `yaml-cpp: error at line 1, column 1: bad conversion`

## System (please complete the following information)
 - OS: Ubuntu Jammy
 - ROS 2 Distro: Humble/Rolling
 - Version: apt-version tested on 14.10.22

## Additional context
** Add any other context about the problem here **

MichaelOrlov, 2022-10-15:
@geoeo Could you please attach your QoS yaml which is giving error on playback?

geoeo, 2022-10-15:
This is the vanilla file I tried which is published on the readme page

```
/topic_name:
  history: keep_last
  depth: 10
  reliability: reliable
  durability: volatile
  deadline:
    # unspecified/infinity
    sec: 0
    nsec: 0
  lifespan:
    # unspecified/infinity
    sec: 0
    nsec: 0
  liveliness: system_default
  liveliness_lease_duration:
    # unspecified/infinity
    sec: 0
    nsec: 0
  avoid_ros_namespace_conventions: false
  ```
  I also tried chaning `/topic_name:` to `/tf_static:` 
  
  Its set via the `offered_qos_profiles` attribute when creating `TopicMetadata` (via python).

MichaelOrlov, 2022-12-06:
@geoeo I've tried to reproduce your issue - and wasn't able to reproduce it.
I checked out latest Humble branch compiled it recorded new bag file with default qos settings from `ros2 run demo_nodes_cpp chatter`.
Have tried to replay it via `ros2 bag play --qos-profile-overrides-path override.yaml test_qos ` and it's played back successfully.
I don't see any errors with messages like "yaml-cpp: error at line 1, column 1: bad conversion".
I've tried to change the first line in override.yaml from `/topic_name:` to the `/chatter:` - still no errors.

You mentioned that `Its set via the offered_qos_profiles attribute when creating TopicMetadata (via python).`
Can you please clarify what does it means? How are you replaying your recorded bag?

MichaelOrlov, 2023-02-01:
@geoeo If you are using Python interface there are should be corresponding enum values instead of the names for `reliability` and `durability` fields.   
This issue looks like duplicate for #1183 and I've provided example in https://github.com/ros2/rosbag2/issues/1183#issuecomment-1341790030 how to properly set values for `offered_qos_profiles` attribute in python.

MichaelOrlov, 2023-02-01:
- Closing as duplicate for #1183. 
- Feel free to reopen if it's not true and issue is still reproducible.

closed 2023-02-01 by MichaelOrlov