ros2/rosbag2 #1129: Problem playing bags files stored online

ghanim-mukhtar, 2022-10-18:
## Description
I have bag files that are stored in GCloud buckets and I mount the whole folder into my PC via `gcsfuse` ([see here](https://github.com/GoogleCloudPlatform/gcsfuse/blob/master/docs/installing.md)) but then when I try to play them I have a lot of problems and the node hangs a lot the topics take for ever to display and eventually the whole thing freezes and the mounter bucket is lost. 

### Expected Behavior
I expected that the ROS2 Bag play will treat the mounted bucket as if a folder is in the PC and the playing will go smoothly and I can visualize all topics on RVIZ normally. 

### Actual Behavior
As said before, using (--read-ahead-queue-size of 100), the reading starts but the rendering of topics is very slow, and after a while all freezes and I loose the mounted bucket. 

## To Reproduce
** Steps to reproduce the behavior, e.g.
1. Generate large rosbags (I have folders with several GB like 5GB for each) and store them in a GCloud bucket (I cannot share the bucket because it is private thing and  I don't have right to reproduce it).
2. Mount the bucket using gcsfuse (`gcsfuse --implicit-dirs your-bucket /local/mounting/point`).
3. Then you can play the bag and see the performance (I use RVIZ2). 

## System (please complete the following information)
 - OS: [Ubuntu focal]
 - ROS 2 Distro: [Foxy]

MichaelOrlov, 2022-10-19:
@ghanim-mukhtar A lot of things got changed since `Foxy`.
Can you pleas try to reproduce this issue with `Rolling`?

Another question: 
Do you see message like `"Message queue starved. Messages will be delayed. Consider increasing the --read-ahead-queue-size option."` when playing your bag?

james-rms, 2022-10-21:
@ghanim-mukhtar some possible workarounds you can try:

1. convert your bags to MCAP using the [mcap CLI tool](https://mcap.dev/getting-started/ros-2.html#using-the-mcap-cli-tool) and the use the [MCAP storage plugin](https://github.com/ros-tooling/rosbag2_storage_mcap) to play back your bags. MCAP is designed to be efficient to read over a network connection, and should perform a little better than SQLite3 here.
2. Convert your bags to MCAP, then instead of using RViz + ros2 bag + gcsfuse, use `gsutil signurl` to create signed https links to your bags and view them using [Foxglove Studio](https://studio.foxglove.dev/). This should be even more efficient since it skips the FUSE layer entirely.

ghanim-mukhtar, 2022-10-26:
@MichaelOrlov so I tried with Rolling and I had even worse performance, the rosbag playing just hangs without anything showing in RVIZ. 
And to answer your question, yes I head the warning `"Message queue starved. Messages will be delayed. Consider increasing the --read-ahead-queue-size option."` when I was reading with `--read-ahead-queue-size 100` under foxy. If I don't use this option it will take forever for the bag file to load. With this option there are topics showing even though they are very slow. 

@james-rms for your two suggestions here is what I got:

1. When I convert a local bag (`.db3`) to (`.mcap`) file then upload this file to cloud and replay it (using RViz + ros2 bag + gcsfuse), it performed very well. The problem is afterwards when I try to do the conversion on the online files, so I mounted the bucket on my PC and did the same command but it hangs for a long time and I don't how to overcome that.
2. For the second solution, I had a problem that my bucket is private not public, and even though I have created an  authenticated URL I still wasn't able to play it on Foxglove I got this error `Error initializing: Fetching remote file failed. TypeError: NetworkError when attempting to fetch resource. Sometimes this is due to a CORS configuration error on the server. Make sure CORS is enabled.`

james-rms, 2022-10-26:
> Another thing is that I have ros bags that consists of several files (of about 5GB each) so do you know how to convert the whole folder that contains them? Normally they have also a metadata.yaml file associated with them.

You can use `ros2 bag convert` with the MCAP storage plugin (in rolling) to do the conversion also - this works on entire folders full of `.db3` files https://mcap.dev/getting-started/ros-2.html#using-ros2-bag-convert

> For the second solution, I had a problem that my bucket is private not public, and even though I have created an authenticated URL I still wasn't able to play it on Foxglove

Did you try the desktop app? https://foxglove.dev/download

MichaelOrlov, 2022-10-27:
@ghanim-mukhtar I would recommend to experiment with `--read-ahead-queue-size` it looks like in your recorded data set configuration 100 is to low and 1000 which is by default is to much.
It depends from the message size.
Consider it as caching 1000 messages from IMU which up to hundred bytes or 1000 messages from camera which is up to several megabytes.
In case of camera you probably will not have enough RAM and it will try to use swap file or will cause a crash. 

Or consider to prune your bags from topics which is out of point of interests via the same `ros2 bag convert` command.

MichaelOrlov, 2023-03-17:
- Closing as stale issue.  
It was provided advise about configuring `--read-ahead-queue-size` but no response from author.
Feel free to reopen if this issue still needs to be addressed.

closed 2023-03-17 by MichaelOrlov