ros2/rosbag2 #1134: Make `ros2bag` CLI help text and descriptions plugin-agnostic

james-rms, 2022-10-21:
## Description
Right now the help text for several arguments refers directly to `sqlite3`-specific features, for example:

1. This description of storage preset profiles: https://github.com/ros2/rosbag2/blob/rolling/ros2bag/ros2bag/verb/record.py#L151
2. This description of storage options: https://github.com/ros2/rosbag2/blob/rolling/ros2bag/ros2bag/verb/record.py#L170
3. This set of choices for storage preset profile: https://github.com/ros2/rosbag2/blob/19158f6debe25640507881f9a2c01cdb1fe4741b/ros2bag/ros2bag/verb/record.py#L149

This leads to an awkward experience for anyone using a non-SQLite3 storage plugin. The most pressing issue is that the storage profile preset validation in `record.py` precludes any plugin author from creating storage preset profiles with names other than `resilient` and `none`.

## Related Issues

Relates to #1116 

## Completion Criteria

Complete when:
1. the `ros2 bag` CLI help text output either depends on the storage plugin selected, or points the user towards documentation specific to their storage plugin.
2. Plugin authors can define storage preset profiles other than "none" and "resilient".

## Implementation Notes / Suggestions

## Testing Notes / Suggestions

Manual testing should be sufficient for this change.
** All features in this project need tests. Please give some input on cases that will need to be tested - and how the testing might be implemented. **

emersonknapp, 2022-11-01:
Might want to  apply this to README as well, providing a sub-README in the storage plugin with its details - removing anything sqlite specific from main README

MichaelOrlov, 2023-02-01:
@james-rms @emersonknapp Can we close this issue as resolved in #1209 ?

emersonknapp, 2023-02-01:
Yes, resolved in #1209

closed 2023-02-01 by emersonknapp