ros2/rosbag2 #1137: Writing a rosbag

imouta, 2022-10-25:
Hello,

I'm new to ROS2 and I've been trying to use it to read messages from rosbag2 play and write them into a new rosbag in C++. 
I started with the code below :
[https://docs.ros.org/en/foxy/Tutorials/Advanced/Recording-A-Bag-From-Your-Own-Node-CPP.html](url)
However, it seems that topics are not created and serialized messages are not read.

What could be the problem?

I appreciate any help.

MichaelOrlov, 2022-10-25:
Hi @imouta. I would recommend to use `ros2 bag convert` CLI option instead.  
Please refer to the https://github.com/ros2/rosbag2#converting-bags for details.

MichaelOrlov, 2022-10-25:
- Closing as it doesn't relates to the rosbag2 issue.
 @imouta Please ask questions in https://answers.ros.org/
Rule of thumb to use issue tracker for real issues in rosbag2.

Please feel free to reopen this issue if you think it's really a problem in `rosbag2`.

closed 2022-10-25 by MichaelOrlov