ros2/rosbag2 #1139: "Hidden topics are not recorded" Spam

tonynajjar, 2022-10-25:
## Description
** clear and concise description of the bug is **

### Expected Behavior
** clear and concise description of what you _expected_ to happen **

### Actual Behavior
** clear and concise description of what _actually_ happened. include copied logs whenever possible **

## To Reproduce
** Steps to reproduce the behavior, e.g.
1. ros2 bag record -a

## System (please complete the following information)
 - OS: Ubuntu Jammy
 - ROS 2 Distro: Humble
 - Version: humble branch

## Additional context
```
root@logi-XMG-CORE-REN-M20:/code/ros2_ws/src# ros2 bag record -a
[INFO] [1666731856.970550246] [rosbag2_recorder]: Press SPACE for pausing/resuming
[INFO] [1666731856.971607274] [rosbag2_storage]: Opened database 'rosbag2_2022_10_25-23_04_16/rosbag2_2022_10_25-23_04_16_0.db3' for READ_WRITE.
[INFO] [1666731856.971671794] [rosbag2_recorder]: Event publisher thread: Starting
[INFO] [1666731856.971853329] [rosbag2_recorder]: Listening for topics...
[WARN] [1666731856.972882483] [ROSBAG2_TRANSPORT]: Hidden topics are not recorded. Enable them with --include-hidden-topics
[WARN] [1666731856.973172571] [ROSBAG2_TRANSPORT]: Hidden topics are not recorded. Enable them with --include-hidden-topics
[WARN] [1666731856.975225156] [ROSBAG2_TRANSPORT]: Hidden topics are not recorded. Enable them with --include-hidden-topics
[WARN] [1666731856.975483405] [ROSBAG2_TRANSPORT]: Hidden topics are not recorded. Enable them with --include-hidden-topics
[WARN] [1666731856.975685965] [ROSBAG2_TRANSPORT]: Hidden topics are not recorded. Enable them with --include-hidden-topics
[WARN] [1666731856.975916244] [ROSBAG2_TRANSPORT]: Hidden topics are not recorded. Enable them with --include-hidden-topics
[INFO] [1666731856.979204095] [rosbag2_recorder]: Subscribed to topic '/waypoints'
[INFO] [1666731856.980079495] [rosbag2_recorder]: Subscribed to topic '/rosout'
[INFO] [1666731856.980759173] [rosbag2_recorder]: Subscribed to topic '/parameter_events'
[INFO] [1666731856.981313085] [rosbag2_recorder]: Subscribed to topic '/initialpose'
[INFO] [1666731856.981671876] [rosbag2_recorder]: Subscribed to topic '/events/write_split'
[INFO] [1666731856.982128645] [rosbag2_recorder]: Subscribed to topic '/clicked_point'
[WARN] [1666731856.983047054] [ROSBAG2_TRANSPORT]: Hidden topics are not recorded. Enable them with --include-hidden-topics
[WARN] [1666731856.983295815] [ROSBAG2_TRANSPORT]: Hidden topics are not recorded. Enable them with --include-hidden-topics
[WARN] [1666731856.985323565] [ROSBAG2_TRANSPORT]: Hidden topics are not recorded. Enable them with --include-hidden-topics
[WARN] [1666731856.985568652] [ROSBAG2_TRANSPORT]: Hidden topics are not recorded. Enable them with --include-hidden-topics
[WARN] [1666731856.985758833] [ROSBAG2_TRANSPORT]: Hidden topics are not recorded. Enable them with --include-hidden-topics
[WARN] [1666731856.985981236] [ROSBAG2_TRANSPORT]: Hidden topics are not recorded. Enable them with --include-hidden-topics
[WARN] [1666731857.090086053] [ROSBAG2_TRANSPORT]: Hidden topics are not recorded. Enable them with --include-hidden-topics
[WARN] [1666731857.090398905] [ROSBAG2_TRANSPORT]: Hidden topics are not recorded. Enable them with --include-hidden-topics
[WARN] [1666731857.092402148] [ROSBAG2_TRANSPORT]: Hidden topics are not recorded. Enable them with --include-hidden-topics
[WARN] [1666731857.092620677] [ROSBAG2_TRANSPORT]: Hidden topics are not recorded. Enable them with --include-hidden-topics
[WARN] [1666731857.092807607] [ROSBAG2_TRANSPORT]: Hidden topics are not recorded. Enable them with --include-hidden-topics
[WARN] [1666731857.093056478] [ROSBAG2_TRANSPORT]: Hidden topics are not recorded. Enable them with --include-hidden-topics
[WARN] [1666731857.196338474] [ROSBAG2_TRANSPORT]: Hidden topics are not recorded. Enable them with --include-hidden-topics
[WARN] [1666731857.196660304] [ROSBAG2_TRANSPORT]: Hidden topics are not recorded. Enable them with --include-hidden-topics
[WARN] [1666731857.198623338] [ROSBAG2_TRANSPORT]: Hidden topics are not recorded. Enable them with --include-hidden-topics
[WARN] [1666731857.198841470] [ROSBAG2_TRANSPORT]: Hidden topics are not recorded. Enable them with --include-hidden-topics
[WARN] [1666731857.199075528] [ROSBAG2_TRANSPORT]: Hidden topics are not recorded. Enable them with --include-hidden-topics
[WARN] [1666731857.199375941] [ROSBAG2_TRANSPORT]: Hidden topics are not recorded. Enable them with --include-hidden-topics
[WARN] [1666731857.302802847] [ROSBAG2_TRANSPORT]: Hidden topics are not recorded. Enable them with --include-hidden-topics
[WARN] [1666731857.303082445] [ROSBAG2_TRANSPORT]: Hidden topics are not recorded. Enable them with --include-hidden-topics
[WARN] [1666731857.305119559] [ROSBAG2_TRANSPORT]: Hidden topics are not recorded. Enable them with --include-hidden-topics
[WARN] [1666731857.305389354] [ROSBAG2_TRANSPORT]: Hidden topics are not recorded. Enable them with --include-hidden-topics
[WARN] [1666731857.305628586] [ROSBAG2_TRANSPORT]: Hidden topics are not recorded. Enable them with --include-hidden-topics
[WARN] [1666731857.305858167] [ROSBAG2_TRANSPORT]: Hidden topics are not recorded. Enable them with --include-hidden-topics
[WARN] [1666731857.409153814] [ROSBAG2_TRANSPORT]: Hidden topics are not recorded. Enable them with --include-hidden-topics
[WARN] [1666731857.409496091] [ROSBAG2_TRANSPORT]: Hidden topics are not recorded. Enable them with --include-hidden-topics
[WARN] [1666731857.411587770] [ROSBAG2_TRANSPORT]: Hidden topics are not recorded. Enable them with --include-hidden-topics
[WARN] [1666731857.411888798] [ROSBAG2_TRANSPORT]: Hidden topics are not recorded. Enable them with --include-hidden-topics
[WARN] [1666731857.412126329] [ROSBAG2_TRANSPORT]: Hidden topics are not recorded. Enable them with --include-hidden-topics
[WARN] [1666731857.412386401] [ROSBAG2_TRANSPORT]: Hidden topics are not recorded. Enable them with --include-hidden-topics
[INFO] [1666731857.424893677] [rclcpp]: signal_handler(signum=2)
[INFO] [1666731857.517842777] [rosbag2_recorder]: Event publisher thread: Exiting
[INFO] [1666731857.589452913] [rosbag2_cpp]: Writing remaining messages from cache to the bag. It may take a while
```

The spam cannot be intended right?

rshanor, 2022-11-09:
I was also annoyed by this, on my fork but changing that print to `RCUTILS_LOG_WARN_THROTTLE_NAMED(RCUTILS_STEADY_TIME, 10000, ROSBAG2_TRANSPORT_PACKAGE_NAME, "Hidden topics are not recorded. Enable them with --include-hidden-topics");`. Happy to push this upstream if others are OK with the change...

MichaelOrlov, 2022-11-09:
@rshanor Yes, your solution looks legitimate. 
It would be highly appreciated if you will create PR for it.

MichaelOrlov, 2022-11-15:
Agree make sense to use log once primitive.
```
RCUTILS_LOG_WARN_ONCE_NAMED(
      ROSBAG2_TRANSPORT_PACKAGE_NAME,
      "Hidden topics are not recorded. Enable them with --include-hidden-topics");
```

closed 2022-12-28 by emersonknapp

russkel, 2023-05-04:
Any chance this could be backported to humble?

tonynajjar, 2023-05-04:
+1 to backporting to humble

MichaelOrlov, 2023-05-04:
Already WIP https://github.com/ros2/rosbag2/pull/1323