ros2/rosbag2 #1143: Add ROS_DISTRO to `.db3` bags

james-rms, 2022-10-31:
## Description

This gives tooling some ability to determine the context of a bag without the original ros workspace it was recorded in.

## Related Issues

## Completion Criteria

* Complete when `.db3` bags contain some table with a row containing the ROS distro the bag was recorded with.
* One idea would be to put the distro in the `metadata.yaml` file, but we should be pushing towards bag files being self-contained where possible.

## Implementation Notes / Suggestions

* Add something to the plugin API to pass the distro into the plugin (OR have the plugin directly inspect the environment variable)
* The plugin API is a better place for this because it communicates the expectation to future storage plugins

## Testing Notes / Suggestions

* should be reasonably simple to test with the current set of storage plugin tests.

emersonknapp, 2022-10-31:
Metadata is (mostly) redundantly stored within storage files - so we should store it _both_ in metadata and the individual file

emersonknapp, 2022-11-01:
Closing as duplicate of #1086

closed 2022-11-01 by emersonknapp

MichaelOrlov, 2022-11-02:
@james-rms We do have open issue
-  #1108 and
-  #1109 
Which should satisfy all possible needs. It doesn't make sense to store `ROS_DISTRO` in addition to the DB schema versioning.
At least I am not envision any cases when it would be useful. 
Please note that DB schema could be changed somewhere in between ROS2 distros and releases.