ros2/rosbag2 #1152: sensor_msgs/PointCloud2 messages are lost when using ros2 bag play

maghauke, 2022-11-08:
## Description
I have a ros2 bag with lidar messages (sensor_msgs/PointCloud2), they are published at 10 Hz with the size of approximately 3 MB each. To validate that all the data is being played I use: ros2 topic hz /lidar/point. The output from this command shows that some messages are lost, and I dont understand why.

### Expected Behavior
The expected behaviour is that all messages are received by the ros2 topic hz subscriber.

### Actual Behavior
The actual behaviour is some messages are not received by the ros2 topic hz subscriber. In my bag I have 370 sensor_msgs/PointCloud2 messages, and only 320 of them are actually received.

## To Reproduce
- Open a terminal and ros2 topic hz /pointcloud2/topic
- Open a new terminal and play a ros2 bag with the subscribed pointcloud2 message: ros2 bag play pointcloud_bag/
 - Compare the actual number of pointcloud2 messages on the /pointcloud2/topic using the metadata.yaml file with the last printed "window" size after the bag is finished. These should be the same. 
 - Another check is to read of the "max" dt, which should be not much more than the expected 1/frequency.

## System (please complete the following information)
 - OS: Ubuntu 22.04 in docker container
 - ROS 2 Distro: Humble
 - Version: perception

## Additional context
I have also tested this on a bare-metal ubuntu 22.04 laptop, where the problem is even worse. Here ros2 topic hz, only receives 1/10 messages. I have also set up an programmatic ros2 bag reader to see that all 370 messages are possible to read, which they are.

MichaelOrlov, 2022-11-08:
@maghauke Likely the problem in the QoS settings.  Pleas try to verify this problem with overridden QoS settings for reliable and transient local settings. 
Also make sure to have enough queue size on receiver side. 
`ros2 topic hz` is not the best solution for verification.

MichaelOrlov, 2023-02-01:
- Closing as stale since no response for some time. Please feel free to reopen if think that this is still an actual issue and overriding QoS settings doesn't help.

closed 2023-02-01 by MichaelOrlov