ros2/rosbag2 #1157: this command is not available：$ ros2 bag record -a -d 9000

SparkCK, 2022-11-10:
I want to record the bag for a fixed length of time.But the following commands always report errors.
'''
$ ros2 bag record -a -d 9000
'''
The error is：
'''
usage: ros2 [-h] Call `ros2 <command> -h` for more detailed usage. ...
ros2: error: unrecognized arguments: -d
'''
[image]

MichaelOrlov, 2022-11-10:
@SparkCK I've just verified it on `Rolling` ROS2 distro and it works for me
```
morlov@ros2_rolling:~/ros2_rolling/recording$ ros2 bag record -a -d 9000
[INFO] [1668110954.626044377] [rosbag2_recorder]: Press SPACE for pausing/resuming
[INFO] [1668110954.628684440] [rosbag2_storage]: Opened database 'rosbag2_2022_11_10-12_09_14/rosbag2_2022_11_10-12_09_14_0.db3' for READ_WRITE.
[INFO] [1668110954.629461509] [rosbag2_recorder]: Event publisher thread: Starting
[INFO] [1668110954.629772817] [rosbag2_recorder]: Listening for topics...
[INFO] [1668110954.631193022] [rosbag2_recorder]: Subscribed to topic '/rosout'
[INFO] [1668110954.631739230] [rosbag2_recorder]: Subscribed to topic '/events/write_split'
[INFO] [1668110959.156991318] [rclcpp]: signal_handler(signum=2)
[INFO] [1668110959.168364599] [rosbag2_recorder]: Event publisher thread: Exiting
[INFO] [1668110959.200645233] [rosbag2_cpp]: Writing remaining messages from cache to the bag. It may take a while
morlov@ros2_rolling:~/ros2_rolling/recording$ 
```
What ROS2 distro or branch are you using?

SparkCK, 2022-11-11:
>

SparkCK, 2022-11-11:
My system is Ubuntu 20.04. ROS2 version is foxy。It still does not work.Thanks

MichaelOrlov, 2022-11-11:
@SparkCK Now it's became clear. It doesn't work because this feature was recently developed and is not present in `foxy` distro.

SparkCK, 2022-11-14:
Thanks very much .

closed 2022-11-14 by SparkCK