ros2/rosbag2 #1158: Autodetect MCAP in rosbag2_cpp open

doisyg, 2022-11-11:
## Description

Ongoing discussion with @clalancette here: https://github.com/ros2/cartographer_ros/pull/67
Should rosbag automatically detects the bag format in the open method here ?
https://github.com/ros2/rosbag2/blob/c611a8010a0aaaba8bff47c42846aae2374fe81f/rosbag2_cpp/include/rosbag2_cpp/reader.hpp#L64-L78

And if yes, how ? Doing it based the file extension ?

Related:
- Same for python api ?
- Same for ros2 bag play cli ? (instead of passing the -s flag)

MichaelOrlov, 2022-11-15:
@doisyg @clalancette MCAP or SQLite3 bags  auto detection has already been done in https://github.com/ros2/rosbag2/pull/1072
It's already works for `ros2 bag play`

As regards to the aforementioned `void open(const std::string & uri);` CPP and Python API - will need to make one line fix
in 
https://github.com/ros2/rosbag2/blob/c611a8010a0aaaba8bff47c42846aae2374fe81f/rosbag2_cpp/src/rosbag2_cpp/reader.cpp#L42
remove entire line with `storage_options.storage_id = "sqlite3";` initialization.
If `storage_options.storage_id` is an empty string - `SequentialReader` will try to auto-detect storage plugin by file extension.

emersonknapp, 2022-11-18:
I just ran into this myself and have a solution in https://github.com/ros2/rosbag2/pull/1167

PR #1167 (2022-11-21): Reader and writer can use default storage by not specifying
Don't hardcode anything for reader, as it can now autodetect.

Allow users to not specify a storage id for writer, in which case it will fall back to the default.

Closes #1158
closed 2022-11-21 by emersonknapp