ros2/rosbag2 #1169: Play back bag file by node

labib-c, 2022-11-19:
## Description

When recording messages sent to a topic and playing it back, it will replay every message sent to the topic (as expected). It would be useful to be able to either record messages or replay messages by **node** for more granular testing. 

## Completion Criteria

* add ability to record node
OR
* add ability to play bagfile by node

If this already exists, please close this issue

MichaelOrlov, 2022-11-20:
Hi @labib-c, does this example https://github.com/ros2/rosbag2/blob/rolling/rosbag2_examples/rosbag2_examples_cpp/src/simple_bag_recorder.cpp satisfy your needs?

MichaelOrlov, 2023-02-01:
@labib-c Or when you say `add ability to play bagfile by node` you meant a situation when during recording there was multiple nodes with their own publishers which was publishing on the same topic and you want to reproduce playback the same way splitting messages on multiple publishers?

emersonknapp, 2023-02-01:
The way I read this is:
* Record only the topics being published by a specified node
* On topics being published by several nodes, only record the messages produced by the specified node, rejecting messages from other publishers

> reproduce playback the same way splitting messages on multiple publishers?

This particular use case is related to https://github.com/ros2/rosbag2/issues/629 - where we would like to recreate the original system publisher configurations by having several playback publishers. I will rewrite that issue to match my latest understanding of what we want to do.

labib-c, 2023-02-01:
@MichaelOrlov @emersonknapp 

> Record only the topics being published by a specified node

> On topics being published by several nodes, only record the messages produced by the specified node, rejecting messages from other publishers

Yes this is what I was requesting, I apologize for the poor wording. I created a personal Python script to perform this behaviour but I believe it would be helpful to include this as part of rosbag itself

MichaelOrlov, 2023-02-02:
The more generic case would be to create separate topic for each publisher and prefixing it with publisher ID or publisher name.
In this case we can easily differentiate publishers and automatically associate messages and QoS settings with it. 
Don't need to delete messages - we can just filter them out either on recording or on replay.

BTW we have already discussed this scenario when we were designing MCAP format specification. 
cc: @defunctzombie @amacneil @emersonknapp

emersonknapp, 2023-02-02:
> we have already discussed this scenario when we were designing MCAP format specification

Yes, and the discussion in #629 started to touch on this even before that, but didn't yet go all the way to request the generic feature set of "record publisher id per message and replicate original publisher layout in playback"

> The more generic case would be to create separate topic for each publisher and prefixing it with publisher ID or publisher name.

Maybe that would match up to an "under the hood" implementation, but I don't think it would be the user-facing API.

In MCAP, for example, we would probably create separate `Channel` instances per publisher, with the same `Channel.topic` name, but differentiating information like QoS profile and publisher ID in `Channel.metadata`. Making a distinction between Channel and Topic as separate concepts here. In `sqlite3` perhaps we would have a new column in table `messages`, so that it was `id, topic_id, publisher_id, timestamp, data`. Or maybe it would be implemented as separate `topic_ids` with an illegal-character delimiter for the prefix.