ros2/rosbag2 #1172: zstd_vendor breaks CMake when cross-compiling for QNX

jpsamper2009, 2022-11-21:
## Description

- When cross-compiling `rosbag2` for x86_64 QNX, `zstd_vendor` will download and compile `libzstd.so`, then copy it to the install directory. `libzstd.so` is a dependency of the CMake executable on the host, so on the next `cmake` invocation, the runtime linker finds the QNX version of `libzstd.so`, causing `cmake` to fail
- The root cause is that `colcon` invokes `LD_LIBRARY_PATH=/path/to/workspace/install/lib:$LD_LIBRARY_PATH cmake ...` in order to make other packages' resources available to the current package; however, this have the side effect of also changing the `LD_LIBRARY_PATH` for the host's CMake.

### Expected Behavior
- Compiling `libzstd.so` for x86_64 QNX does not interfere with the `libzstd.so` need by the `cmake` on the host.

### Actual Behavior
1. `zstd` is not found for x86_64 QNX
2. `zstd_vendor` downloads, compiles, and installs `libzstd.so` in the workspace's install directory
3. `cmake` is invoked for another package, finding the new `libzstd.so`
4. Since `libzstd.so` is compiled for QNX (and not Linux), `cmake` fails to run

## To Reproduce
1. Follow the instructions in https://ros2-qnx-documentation.readthedocs.io/en/galactic/build_from_source.html to compile ROS2 for QNX, except:
    1. In "Building Steps - 5", use `CPU=x86_64`
    2. Modify `build-ros2.sh` to invoke `colcon` twice, the first time with `--packages-up-to zstd_vendor`, the second time with `--packages-select gmock_vendor`: i.e.
    ```
    build () {
        ...
       colcon build ...
            ...
            --packages-up-to zstd_vendor \
            --cmake-args \
                 ....
       colcon build ...
            ...
            --packages-up-to gmock_vendor \
            --cmake-args \
                 ....
        ...
    }
    ```
2. The second `colcon` command fails with something like:
    ```
    /usr/bin/cmake: error while loading shared libraries: libregex.so.1: cannot open shared object file: No such file or directory
    ```

(Side note: The instructions for compiling ROS2 for QNX are broken in other ways, but it was easier to point to those instructions than to explain here all the other changes that are not related to this bug report)

## System
 - Host OS/Arch: Linux x86_64
 - Target OS/Arch: QNX 7.1 x86_64
 - ROS 2 Distro: Galactic, Humble
 - Version: See https://ros2-qnx-documentation.readthedocs.io/en/galactic/build_from_source.html

## Additional context
- I imagine the fix is to follow a similar scheme to `yaml_cpp_vendor` where the libraries are installed elsewhere, and the `_vendor` package makes them available

jpsamper2009, 2022-11-23:
Applying the following patch and renaming `zstd_vendor-extras.cmake` to `zstd_vendor-extras.cmake.in` seems to fix the issue (but I haven't tested if it breaks other builds).

```diff
diff --git a/zstd_vendor/CMakeLists.txt b/zstd_vendor/CMakeLists.txt
index e3d4a22..e682a58 100644
--- a/zstd_vendor/CMakeLists.txt
+++ b/zstd_vendor/CMakeLists.txt
@@ -55,7 +55,7 @@ macro(build_zstd)
 
   install(
     DIRECTORY ${CMAKE_CURRENT_BINARY_DIR}/${PROJECT_NAME}_install/
-    DESTINATION ${CMAKE_INSTALL_PREFIX}
+    DESTINATION ${CMAKE_INSTALL_PREFIX}/opt/zstd_vendor
     USE_SOURCE_PERMISSIONS)
 endmacro()
 
@@ -67,4 +67,4 @@ endif()
 
 install(DIRECTORY cmake DESTINATION share/${PROJECT_NAME})
 
-ament_package(CONFIG_EXTRAS zstd_vendor-extras.cmake)
+ament_package(CONFIG_EXTRAS zstd_vendor-extras.cmake.in)
diff --git a/zstd_vendor/zstd_vendor-extras.cmake b/zstd_vendor/zstd_vendor-extras.cmake
index 07472ac..40287d4 100644
--- a/zstd_vendor/zstd_vendor-extras.cmake
+++ b/zstd_vendor/zstd_vendor-extras.cmake
@@ -1,1 +1,5 @@
+if(NOT DEFINED ENV{zstd_ROOT_DIR})
+  set(zstd_ROOT_DIR "${@PROJECT_NAME@_DIR}/../../../opt/zstd_vendor")
+endif()
+
 list(INSERT CMAKE_MODULE_PATH 0 "${zstd_vendor_DIR}/Modules")
```

emersonknapp, 2022-11-23:
If you propose this change as a pull request, then it would automatically run Ubuntu build and we would run linux-aarch64 and windows build as part of the review process.

"prs welcome" :)

jpsamper2009, 2022-11-23:
@emersonknapp For sure. I just wanted to make sure it is the correct approach before going through the effort 😅

chachoi, 2024-04-16:
I know this is an old issue, but I found a "fix".

Simply set `LD_PRELOAD` to `libzstd.so`'s absolute path.

For me it's:
`export LD_PRELOAD=$LD_PRELOAD:/usr/lib/x86_64-linux-gnu/libzstd.so`.

If you set `LD_PRELOAD`, your PC would look for the `libzstd.so` in there first, rather than the `libzstd.so` which is compiled for QNX.

emersonknapp, 2025-02-21:
Looks like we have a workaround, and the original request is quite stale - I'm going to close this for now, please feel free to reopen it if an actionable issue remains.

closed 2025-02-21 by emersonknapp