ros2/rosbag2 #1182: Improve cache interface

YXL76, 2022-11-27:
## Description

Current `consume_callback_` needs a `vector` input.

https://github.com/ros2/rosbag2/blob/86efa58dca1945bb24055a5a109929c186f38903/rosbag2_cpp/include/rosbag2_cpp/cache/cache_consumer.hpp#L65-L66

`MessageCacheCircularBuffer` will copy data from `deque` to `vector`, which is redundant.

https://github.com/ros2/rosbag2/blob/86efa58dca1945bb24055a5a109929c186f38903/rosbag2_cpp/src/rosbag2_cpp/cache/message_cache_circular_buffer.cpp#L64-L70

## Suggestions

Convert it to iterator.

MichaelOrlov, 2022-12-01:
As always PRs are welcome.

MichaelOrlov, 2023-03-17:
After some consideration I have to decline this request. And there are 2 reasons for that:
1. There are no actual data copy there. We are copying shared pointers to the serialized buffers. Therefore no burden on performance.
2. Changing interface for callback will involve changes in underlying storage plugins interfaces since we are binding call back to the SequentialWriter::write_messages(..) method
https://github.com/ros2/rosbag2/blob/76665ad8b977721c059b95d307697b8424fed07d/rosbag2_cpp/src/rosbag2_cpp/writers/sequential_writer.cpp#L416-L423
Which is calling `storage_->write(messages);` 
It doesn't worth it to heavily change API on many layers and packages if no marginal improvements in it.

closed 2023-03-17 by MichaelOrlov